greatscottgadgets/ubertooth #251: Time inconsistency in ubertooth host tools

ismaia, 2017-02-02:
Last week I did three times a six hours test with my ubertooth:  
```
time ubertooth-rx -s -t21600 -r bt.cap
```

Analysing collected bluetooth data in wireshark,  the duration of the whole test start at the good time but finishes one hour and half after real time

Still analysing stdout of ubertooth-rx command which is showing local host time, the whole test duration is consistent.

Looking at ubertooth_callback.c, I found nowns variable:
*uint64_t nowns = now_ns_from_clk100ns( ut, rx )* ,  which seems to be the internal ubertooth clock

and  *systime = time(NULL)* which is the host and consistent clock.

It could be a good idea to have an option to save host time in pcap files instead of ubertooth time.

I think that replacing systime by nowns in *_pcap_append_packet* function calls could do the trick

smbeyer8, 2017-08-08:
I am seeing this issue on the latest firmware (2017-03-R2 (API:1.02)). I see timing drift on one ubertooth and timing issues among uberteeth when starting three at the same time (i.e. each ubertooth listening on a different advertisement channel starting at the same time). Is there a short term fix to this? Can I go in and update the firmware? I am currently working on a project that requires all sniffers to be synced with a WiFi sniffer so timing is vital. I've included a capture as an example. This capture is a merging of three ubertooth captures that all started at the exact same time (one for each advertising channel). Thanks for the help and great support! 
[multi_cap_7Aug17.pcap.zip](https://github.com/greatscottgadgets/ubertooth/files/1208911/multi_cap_7Aug17.pcap.zip)

smbeyer8, 2017-08-08:
Just a quick update: I took what @ismaia said and ran with it. I modified the _ubertooth_callback.c_ file to write the system time to the pcap rather than the ubertooth built-in clock. I tried to do it by changing the least amount of code possible. I inserted the following code among the existing code and changed out `nowns` for `sys_time`:
```c
struct timespec time_spec;
clock_gettime(CLOCK_REALTIME, &time_spec);
uint64_t sys_time = time_spec.tv_sec*1000000000ull+time_spec.tv_nsec;

if (ut->h_pcap_le) {
	/* only one of these two will succeed, depending on
	 * whether PCAP was opened with DLT_PPI or not */
	lell_pcap_append_packet(ut->h_pcap_le, sys_time,
			sig, noise,
			refAA, pkt);
	// read the above comment: this function may silently fail
	lell_pcap_append_ppi_packet(ut->h_pcap_le, sys_time,
	                rx->clkn_high,
		        rx->rssi_min, rx->rssi_max,
		        rx->rssi_avg, rx->rssi_count,
		        pkt);
	}
if (ut->h_pcapng_le) {
	lell_pcapng_append_packet(ut->h_pcapng_le, sys_time,
		        sig, noise,
		        refAA, pkt);
}
```
I have only done a little testing, but it appears when `nowns` drifts, `sys_time` is still accurate. I think that this should only be a temporary fix, however, and am not sure what is causing the issue with `nowns`. I uploaded an image of a capture prior to my changes to show that `nowns` will drift for one packet at times and then be on-time again for the next packet. Hopefully this will help getting us closer to a fix if it isn't fixed alreaady.
[ubertooth timing.pdf](https://github.com/greatscottgadgets/ubertooth/files/1209854/ubertooth.timing.pdf)

mikeryan, 2017-08-09:
@sgript your PCAP contains no pairing request. Refer to the FAQ:

https://github.com/mikeryan/crackle/blob/master/FAQ.md#crackle-is-complaining-about-missing-packets-why-cant-i-crack

@Kay887 can you attach a PCAP?

GimliSonOfGloin, 2017-08-10:
@mikeryan, not sure if your post was meant for this Issue? I do not see any of those people or topics? Sorry, just a little confused!

mikeryan, 2017-08-10:
Sorry about that, wrong thread!

commit 9ecd3653 (2020-12-19): host: correctly calculate nanosecond timestamp in LE mode

This was causing time discontinuities of around 2^32 units of 100 ns in
generated PCAP files.

Fixes #202
Fixes #251
closed 2020-12-19 by mikeryan