greatscottgadgets/ubertooth #252: libUSB Error: Command Error: (-9)

cjhermann, 2017-02-06:
### Steps to reproduce
1. ubertooth-rx
2.
3.

### Expected behaviour
I should see a list of LAPs that the ubertooth has detected.

### Actual behaviour
libUSB Error: Command Error: (-9)
Ubertooth running very old firmware found.
Please upgrade to latest released firmware.

### Version information
**Operating system**:
Kali Linux VM

**Ubertooth tools version (ubertooth-rx -V):**
libubertooth 0.3 (3612176)

**libbtbb version:**
 libbtbb 0.4 (9850be8)

**Ubertooth firmware version (ubertooth-util -v):**
2015-10-R1

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

I am getting an error about firmware being old. If I am not mistaken I believe I have the latest firmware so I'm confused as to what the issue is.

dominicgs, 2017-02-06:
You are using the last released firmware but pre-release host code from git.  These are no longer compatible as the interface between the two has changed.  You will either need to upgrade your firmware by building it from the git repository or downgrade your release to 2015-10-R1.

cjhermann, 2017-02-07:
Sorry I am not very familiar with git, so I am still confused. I followed the build from git guide on the wiki. What exactly do I need to update/change? Isn't the release from the build from git guide the 2015-10-R1 release? Sorry again, these might be dumb questions but I'm am very new to git and github.

dominicgs, 2017-02-07:
The [Build Guide](https://github.com/greatscottgadgets/ubertooth/wiki/Build-Guide) gives instructions to build the latest release, which is currently 2015-10-R1.

closed 2020-12-05 by mikeryan