greatscottgadgets/ubertooth #254: Ubertooth One IAR MSVS port for Windows

sevstels, 2017-03-02:
Here released Ubertooth Windows port:  > http://electronics.quantumlah.org
see menu point: >  Ubertooth IAR & MSVS port for Windows

Many small bug`s was removed, also USB transfer bug fixed.
New have high stability operating without hang or crash.
Windows user enjoy...

mikeryan, 2020-12-05:
Thank you (late) for this work! I will consider the best way to merge this in a future release.

sevstels, 2020-12-06:
Glad if you liked it. It works really well under Windows. But it no one needs it anymore. 
The Chinese are massively cloning your project and sale on Aliexpress.
https://aliexpress.ru/af/ubertooth-one.html?d=y&origin=n&SearchText=ubertooth+one&catId=0&initiative_id=SB_20201206020703