greatscottgadgets/ubertooth #255: Ubertooth 'ubertooth' failed to open device ' ': Permission denied

Davedough, 2017-03-03:
### Steps to reproduce
1. Install Ubertooth and Kismet as per the Ubertooth official guide
2. Run kismet as suid user 
3. Load plugin and source.

### Expected behaviour
I'm expecting to see Bluetooth devices scanned in Kismet

### Actual behaviour
It appears the plugin, server and source all loaded correctly, but there's an error line after adding the source that says "Ubertooth 'ubertooth' failed to open device '': Permission denied.

I followed everything in the guide to the letter and my user is in the kismet group, I'm not sure what other permissions are needed. 

### Version information
**Raspbian Jessie 2017-03-02 release**:

**Ubertooth tools version (ubertooth-rx -V):** libubertooth 0.2 (GITDIR-NOTFOUND), libbtbb0.3 (GITDIR-NOTFOUND

**libbtbb version:**

**Ubertooth firmware version (ubertooth-util -v):** Ubertooth 2015-10-R1 (Oddly, I needed to sudo to get this information, which I don't think I've had to before)

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here. Kismet

### Output
```
Insert any commandline or build output here
```

Davedough, 2017-03-03:
Forgot to add the output of kismet_server -c ubertooth

pi@raspberrypi:~ $ kismet_server -c ubertooth
INFO: Not running as root - will try to launch root control binary (/usr/lo
      cal/bin/kismet_capture) to control cards.
INFO: Started kismet_capture control binary successfully, pid 1100
INFO: Reading from config file /usr/local/etc/kismet.conf
debug - 1099 - child creating ipc fdfd
INFO: No 'dronelisten' config line and no command line drone-listen 
      argument given, Kismet drone server will not be enabled.
INFO: Created alert tracker...
INFO: Creating device tracker...
INFO: Registered 80211 PHY as id 0
ERROR: Failed to open user plugin directory (/home/pi/.kismet//plugins/): 
       No such file or directory
INFO: Loaded info for plugin 'spectool_net.so': Plugin name: 'SPECTOOL' 
      Plugin version: '2013-03-R0' Plugin description: 'Spectool-Net'
INFO: Loaded info for plugin 'ubertooth.so': Plugin name: 'UBERTOOTH' 
      Plugin version: '2013-03-R0' Plugin description: 'Ubertooth plugin 
      for Bluetooth baseband protocol'
INFO: Pcap logging for type pcapbtbb
INFO: Activated plugin '/usr/local/lib/kismet/ubertooth.so': 'UBERTOOTH' 
      '2013-03-R0' 
INFO: Loaded info for plugin 'alertsyslog.so': Plugin name: 'ALERTSYSLOG' 
      Plugin version: '2013-03-R0' Plugin description: 'ALERTSYSLOG Plugin'
INFO: Activated plugin '/usr/local/lib/kismet/alertsyslog.so': 
      'ALERTSYSLOG' '2013-03-R0' 
INFO: Loaded info for plugin 'btscan.so': Plugin name: 'BTSCAN' Plugin 
      version: '2013-03-R0' Plugin description: 'Active Bluetooth scanning 
      plugin'
INFO: Activated plugin '/usr/local/lib/kismet/btscan.so': 'BTSCAN' 
      '2013-03-R0' 
INFO: Loaded info for plugin 'kismet-syslog.so': Plugin name: 'SYSLOG' 
      Plugin version: '2013-03-R0' Plugin description: 'SYSLOG Plugin'
ERROR: Kismet-Syslog not redirecting any messages to syslog because no 
       syslogtype= directive found in kismet.conf
INFO: Kismet-Syslog redirecting no messages to syslog
INFO: Activated plugin '/usr/local/lib/kismet/kismet-syslog.so': 'SYSLOG' 
      '2013-03-R0' 
INFO: Kismet will spend extra time on channels 1,6,11
INFO: Kismet will attempt to hop channels at 3 channels per second unless 
      overridden by source-specific options
INFO: Matched source type 'ubertooth' for auto-type source 'ubertooth'
INFO: Using default channel list 'UBERTOOTH' on source 'ubertooth'
INFO: Ubertooth using first USB device that looks like a Ubertooth
INFO: Created source ubertooth with UUID 
      a38448f6-004b-11e7-ae48-b8072e49dc03
INFO: Disabling channel hopping on source 'ubertooth' because it is not 
      capable of setting the channel.
INFO: Will attempt to reopen on source 'ubertooth' if there are errors
INFO: Created TCP listener on port 2501
INFO: Kismet drone framework disabled, drone will not be activated.
INFO: Inserting basic packet dissectors...
INFO: Allowing Kismet frontends to view WEP keys
INFO: Starting GPS components...
INFO: Enabling reconnection to the GPS device if the link is lost
INFO: Using GPSD server on localhost:2947
ERROR: Could not open OUI file '/etc/manuf': No such file or directory
ERROR: Could not open OUI file '/usr/share/wireshark/wireshark/manuf': No 
       such file or directory
ERROR: Could not open OUI file '/usr/share/wireshark/manuf': No such file 
       or directory
ERROR: Could not open OUI file '/Applications/Wireshark.app/Contents/Resour
       ces/share/wireshark/manuf': No such file or directory
ERROR: No OUI files were available, will not resolve manufacturer names 
       for MAC addresses
INFO: Creating network tracker...
INFO: Creating channel tracker...
INFO: Registering dumpfiles...
INFO: Pcap log in PPI format
INFO: Opened pcapdump log file 'Kismet-20170303-19-57-32-1.pcapdump'
INFO: Opened netxml log file 'Kismet-20170303-19-57-32-1.netxml'
INFO: Opened nettxt log file 'Kismet-20170303-19-57-32-1.nettxt'
INFO: Opened gpsxml log file 'Kismet-20170303-19-57-32-1.gpsxml'
INFO: Opened alert log file 'Kismet-20170303-19-57-32-1.alert'
INFO: No spectools= line in config file, will not try to use spectools for 
      spectrum data
INFO: Activated plugin '/usr/local/lib/kismet/spectool_net.so': 'SPECTOOL' 
      '2013-03-R0' 
INFO: Kismet starting to gather packets
libUSB Error: Insufficient Permissions:  (-3)
could not open Ubertooth device
ERROR: Ubertooth 'ubertooth' failed to open device '': Permission denied
INFO: Started source 'ubertooth'
INFO: kismet_capture pid 1100 synced with Kismet server, starting service 
      loop

Davedough, 2017-03-03:
Disregard. I panicked and posted before digging deeper.

`sudo adduser pi plugdev` took care of my issue.

dominicgs, 2017-03-04:
@Davedough Thanks for adding the solution.  I know this is one that can trip people up fairly often.  I'm glad you got it resolved.

closed 2017-03-04 by dominicgs