greatscottgadgets/ubertooth #256: Errors running make in plugin-ubertooth directory

atticusliu, 2017-03-10:
### Steps to reproduce
1. Change directories to /ubertooth/host/kismet/plugin-ubertooth

### Expected behaviour
A proper compilation and linking process should occur as make files do. I'm following the steps specified in the README in /host/kismet/plugin-ubertooth. I apologize if any of this is very vague, I am just starting off. I am more than happy to provide more details when/if necessary.

### Actual behaviour
The make process fails very early on. Specifically after this step:
```
g++ -std=gnu++11 -I./msgpack-1.4.1/include p-thread -fopenmp -Wall -g -O2 -I/usr/src/kismet -I/usr/include/libusb-1.0 -I/usr/local/include/libusb-1.0 -I/usr/include -g -fPIC -c tracker_btbb.cc -o tracker_btbb.o
```
### Version information
**Operating system**:
Linux raspberrypi 4.4.38-v7+
**Ubertooth tools version (ubertooth-rx -V):**
libubertooth 0.2 (GITDIR-NOTFOUND)
**libbtbb version:**
libbtbb 0.3 (GITDIR-NOTFOUND)
**Ubertooth firmware version (ubertooth-util -v):**
2016-10-R1
If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.
Kismet, yes. This is what this issue revolves around.

### Output
```
In file included from tracker_btbb.cc:30:0:
tracker_btbb.h:49:2: error: 'kis_gpa_data' does not name a type
   kis_gpa_data gpsdata;
   ^
tracker_btbb.cc:38:19: error: expected '} before '(' token
   GPS_COMMON_FIELDS(BTBBDEV);
                    ^
tracker_brbb.cc:40:1: error: expected initializer before '}' token
   };
   ^
tracker_btbb.cc:40:1: error: expected declaration before '}' token
Makefile:54: recipe for target 'tracker_btbb.o' failed
make: *** [tracker_btbb.o] Error 1
```

dominicgs, 2017-05-04:
Do you have kismet installed?  Which version?  Do you have the kismet source downloaded too?

mikeryan, 2020-12-05:
Legacy Kismet is no longer supported by this project.

closed 2020-12-05 by mikeryan