greatscottgadgets/ubertooth #257: New Hardware - USB Does not enumerate

dpitchfo, 2017-03-21:
I have created my own Ubertooth One PCB board and having problems with USB not being enumerated.  I have loaded the bootloader via JFLASH with no luck.

### Steps to reproduce
1.  Created the PCB with LPC1756FB processor from Digikey. 
2.  Downloaded and make the ubertooth-2015-10-R1 code following the instructions
https://github.com/greatscottgadgets/ubertooth/wiki/Build-Guide
3.  Make the ./firmware/bootloader and install the bootloader.hex via JFLASH.

### Expected behaviour
I expect the USB to be recognize like the Ubertooth One.  

### Actual behaviour
When I plug the USB in and get dmesg:

>dmesg:
[452484.724070] usb 3-8.1.2: new full-speed USB device number 43 using xhci_hcd
[452484.724196] usb 3-8.1.2: Device not responding to setup address.
[452484.928194] usb 3-8.1.2: Device not responding to setup address.
[452485.132077] usb 3-8.1.2: device not accepting address 43, error -71
[452485.132228] usb 3-8.1-port2: unable to enumerate USB device

>lsusb - doesn't appear. 

### Version information
**Operating system**:
Ubuntu 16.04

**Ubertooth tools version (ubertooth-rx -V):**
ubertooth-rx -V
libubertooth 0.2 (GITDIR-NOTFOUND), libbtbb 0.3 (GITDIR-NOTFOUND)

**libbtbb version:**
Using libbtbb-2015-10-R1

**Ubertooth firmware version (ubertooth-util -v):**
Using ubertooth-2015-10-R1

If you are reporting a problem that involves third party software 
Nope - Just want to get a fresh processor up and running.

### Output
```
Insert any commandline or build output here
```

### Notes
Attached is my schematic.    I did not run out the expansion connector.  I am wondering if I need to short pin 1 & 3 to get into proper mode?    I also used a 20 pin JTAG header since that what I have.   I made my XTAL1 short to ground.   The NXP App note says if you are not using XTAL1, 2, that XTAL1 should be short to gnd and XTAL2 is floating.  (Not sure if this is an issue)

Am I missing something here?  

[uber1.pdf](https://github.com/greatscottgadgets/ubertooth/files/859565/uber1.pdf)

mikeryan, 2017-03-21:
Try shorting the pins on the expansion header and see if it will go into bootloader mode and enumerate that way.

https://github.com/greatscottgadgets/ubertooth/blob/master/host/doc/ubertooth-dfu.md#manually-entering-bootloader

dpitchfo, 2017-03-21:
Thanks for the quick reply:

No luck on connecting Pin 47 of the MCU to GND.  (This is the same as shorting pin 1 & 2 of the expansion header).   Unfortunately I didn't run the expansion header so soldering a wire to a pin is difficult at best.    I will add that to the next respin.  

For the USB, I get the following:
>dmesg
[458909.428054] usb 3-8.1.2: new full-speed USB device number 55 using xhci_hcd
[458909.428177] usb 3-8.1.2: Device not responding to setup address.
[458909.632175] usb 3-8.1.2: Device not responding to setup address.
[458909.836085] usb 3-8.1.2: device not accepting address 55, error -71
[458909.836183] usb 3-8.1-port2: unable to enumerate USB device

Can anyone tell me if I loading the bootloader correctly and using the correct file.   I am pulling the data from /ubertooth-2015-10-R1/firmware/bootloader directory.

1.  make
2.  move the bootloader.hex to windows machine.
3.  Using Segger JFLASH,  Connect the JFLASH, it reads the correct id for the processor.   I load the bootloader.hex and press F5 to write it.   I validate and its written correctly.   

The LEDS that appear are RSTLED (Pin 11 of the MCU) and 1V8LED.   Does RSTLED mean anything?  I do not see "Flashing" of the LEDs.

dpitchfo, 2017-04-04:
All,

I found the issue.    I have to blame it on old age and poor eyesight.   I failed to run GIO1 from the MCU to the CC2400.  This prevented the CC2400 to start up and failed at the boot loading process.   Attached is my final schematic that I recreated in Eagle.  

[UberToothOne.pdf](https://github.com/greatscottgadgets/ubertooth/files/893349/UberToothOne.pdf)

closed 2017-04-10 by dominicgs