greatscottgadgets/ubertooth #259: BTLE Encrypted Connection Incorrect Payload Length

njtrettel, 2017-04-05:
### Steps to reproduce
1. `ubertooth-btle -f -c myFile.pcap`
2. Successfully follow encrypted connection pairing (connection now encrypted)
3. Observe PCAP in wireshark

### Expected behaviour
I expected to see correct packets up until the encryption is set up, and then see a bunch of malformed packets (because of the encryption). The payload lengths of these packets, however, should be correct since the length is not encrypted. 

### Actual behaviour
The packets have correct payload lengths until about 20 packets in. From there on, the lengths get pretty large (biggest payload is 207 bytes), but I can only see around 11 bytes of it. I have three ideas on what could be happening:
 - the ubertooth is for some reason not reading the entire packet
 - the ubertooth is getting the wrong payload length
 - something I'm not aware of in the encrypted connection is causing the length byte to be in a different position than wireshark is expecting. This is doubtful though since the Access Address is reported correctly.
I've seen malformed packets in non-encrypted sessions before that report wildly incorrect values _ocassionally_, but this is happening until the rest of the capture. Not really sure what could be done, but maybe somebody else has an explanation for why?

### Version information
**Operating system**: 
Ubuntu 14.04
**Ubertooth tools version (ubertooth-rx -V)**: 
libubertooth 0.2
**libbtbb version**: 
libbtbb 0.3
**Ubertooth firmware version (ubertooth-util -v)**: 
git-24e7042

## Output
I attached a picture that shows packet 112 with a payload length of 203. The highlighted area of the hex view is the total payload (11 bytes).  
[image]

mikeryan, 2017-04-05:
Can you try to repro with latest firmware and host code (2017-03-R2)

njtrettel, 2017-04-05:
I am in the process of doing that, but had to leave. So I will have to update you later this week or early next week. 

However, I did notice that regardless of the rest of the capture, 9/10 times I am missing one of the handshake values (usually missing either a `confirm` or `random` value), which makes the whole capture useless to me. Is there a maximum data rate worth noting that would cause the ubertooth to skip (unintentionally) these packets? I've tried this on a Raspberry Pi and then with an i7 laptop.

njtrettel, 2017-04-07:
@mikeryan I've been testing with the newest firmware and I'm not able to get a good capture. I'm missing a `random` value every time. I'm using a Raspberry Pi (but have previously tried a laptop) but I'm pretty confident this is not the issue. I'm testing inside a Faraday cage so noise is definitely not the issue either. 

Maybe I should mention (although not extremely relevant to the issue) that I am also using your crackle program, which is why I'm concerned about the length fields (previous picture). Have you ever come across an issue like this where the ubertooth is not receiving some packets?

[image]

mikeryan, 2017-04-10:
Ubertooth routinely misses a lot of packets, although if it's always the same packet that's a little suggestive of a deeper problem.

Are you using a commercial faraday cage? How many times have you attempted to capture the exchange?

njtrettel, 2017-04-26:
@mikeryan Sorry for the delayed response. 

I could still never manage to get a clean capture with all of the values I needed. Then I upgraded to Android 7.0 (had some version of 6 before). Now the ubertooth consistently "sticks" to a few of the data channels. The only cause I could think of is missing a channel map update or something. This may be an anomaly with the device I'm capturing, but I only have one device to test. 

I apologize if this is a known issue. I would look into this further, but I have switched to using a phone with an older version of Android 6 and it seems to be working beautifully again!

closed 2017-04-26 by njtrettel