greatscottgadgets/ubertooth #261: Error building wireshark/plugins/btbb (Build Guide Release 2017-03-R2/ libbtbb 2017-03-R2)

rajpancha1, 2017-04-27:
Hi Team:
I'm new to Project Ubertooth and the board.

I'm installing Ubertooth on freshly imaged Ubuntu 17.04 system (Linux SmartSquareSystems 4.10.0-19-generic #21-Ubuntu SMP Thu Apr 6 17:04:57 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux)

While building wireshark/plugins/btbb I'm getting the following cmake error:
cmake -DCMAKE_INSTALL_LIBDIR=/usr/lib/x86_64-linux-gnu/wireshark/libwireshark3/plugins ..
-- The C compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Plugin will be installed in: /usr/lib/x86_64-linux-gnu/wireshark/libwireshark3/plugins
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for one of the modules 'glib-2.0>=2.4.0'
-- Found GLib2: glib-2.0 /usr/include/glib-2.0;/usr/lib/x86_64-linux-gnu/glib-2.0/include
-- Looking for include file glib/gregex.h
-- Looking for include file glib/gregex.h - not found
-- Looking for include file glib/gchecksum.h
-- Looking for include file glib/gchecksum.h - not found
-- Configuring done
-- Generating done
-- Build files have been written to: /home/rajendra/BT/libbtbb-2017-03-R2/wireshark/plugins/btbb/build

OLD ISSUE: I looked at the old issue https://github.com/greatscottgadgets/ubertooth/issues/36 to see if the same solution applies to the new release also. NCMakeError.log CMakeOutput.logot so, because the FindWireshark.cmake correctly looks for epan/packet.h.

### Steps to reproduce 
1. Followed Build Guide Release 2017-03-R2/ libbtbb 2017-03-R2epan/packet.h

### Expected behaviour
Tell us what you expect should happen

### Actual behaviour
rajendra@SmartSquareSystems:~/BT$ tar xf libbtbb-2017-03-R2.tar.gz
rajendra@SmartSquareSystems:~/BT$ cd libbtbb-2017-03-R2
rajendra@SmartSquareSystems:~/BT/libbtbb-2017-03-R2$ mkdir build
rajendra@SmartSquareSystems:~/BT/libbtbb-2017-03-R2$ cd build
rajendra@SmartSquareSystems:~/BT/libbtbb-2017-03-R2/build$ cmake ..
-- The C compiler identification is GNU 6.3.0
-- The CXX compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Setting version string 2017-03-R2
-- Building shared library
-- Found PythonInterp: /usr/bin/python (found version "2.7.13") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/rajendra/BT/libbtbb-2017-03-R2/build
rajendra@SmartSquareSystems:~/BT/libbtbb-2017-03-R2/build$ make
Scanning dependencies of target btbb
[ 11%] Building C object lib/src/CMakeFiles/btbb.dir/bluetooth_packet.c.o
[ 22%] Building C object lib/src/CMakeFiles/btbb.dir/bluetooth_piconet.c.o
[ 33%] Building C object lib/src/CMakeFiles/btbb.dir/bluetooth_le_packet.c.o
[ 44%] Building C object lib/src/CMakeFiles/btbb.dir/companies.c.o
[ 55%] Building C object lib/src/CMakeFiles/btbb.dir/pcap.c.o
[ 66%] Building C object lib/src/CMakeFiles/btbb.dir/pcapng.c.o
[ 77%] Building C object lib/src/CMakeFiles/btbb.dir/pcapng-bt.c.o
[ 88%] Linking C shared library libbtbb.so
[ 88%] Built target btbb
Scanning dependencies of target btaptap
[ 88%] Built target btaptap
Scanning dependencies of target pcapdump
[100%] Generating build/timestamp
running build
running build_py
creating build
creating build/lib.linux-x86_64-2.7
creating build/lib.linux-x86_64-2.7/pcapdump
copying /home/rajendra/BT/libbtbb-2017-03-R2/python/pcaptools/pcapdump/pcapdump.py -> build/lib.linux-x86_64-2.7/pcapdump
copying /home/rajendra/BT/libbtbb-2017-03-R2/python/pcaptools/pcapdump/__init__.py -> build/lib.linux-x86_64-2.7/pcapdump
[100%] Built target pcapdump
rajendra@SmartSquareSystems:~/BT/libbtbb-2017-03-R2/build$ sudo make install
[sudo] password for rajendra: 
[ 88%] Built target btbb
[ 88%] Built target btaptap
[100%] Built target pcapdump
Install the project...
-- Install configuration: ""
-- Installing: /usr/local/lib/pkgconfig/libbtbb.pc
-- Installing: /usr/local/lib/libbtbb.so.1.0
-- Up-to-date: /usr/local/lib/libbtbb.so.1
-- Up-to-date: /usr/local/lib/libbtbb.so
-- Up-to-date: /usr/local/include/btbb.h
running build
running build_py
running install
running install_lib
running install_egg_info
Removing //usr/local/lib/python2.7/site-packages/pcapdump-0.0.0-py2.7.egg-info
Writing //usr/local/lib/python2.7/site-packages/pcapdump-0.0.0-py2.7.egg-info
-- Up-to-date: /usr/local/bin/btaptap
rajendra@SmartSquareSystems:~/BT/libbtbb-2017-03-R2/build$ 
rajendra@SmartSquareSystems:~/BT/libbtbb-2017-03-R2/build$ sudo apt-get install wireshark wireshark-dev libwireshark-dev cmake
Reading package lists... Done
Building dependency tree       
Reading state information... Done
cmake is already the newest version (3.7.2-1).
libwireshark-dev is already the newest version (2.2.4+gcc3dc1b-1).
wireshark is already the newest version (2.2.4+gcc3dc1b-1).
wireshark-dev is already the newest version (2.2.4+gcc3dc1b-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
rajendra@SmartSquareSystems:~/BT/libbtbb-2017-03-R2/build$ 
rajendra@SmartSquareSystems:~/BT/libbtbb-2017-03-R2/build$ pwd
/home/rajendra/BT/libbtbb-2017-03-R2/build
rajendra@SmartSquareSystems:~/BT/libbtbb-2017-03-R2/build$ cd
rajendra@SmartSquareSystems:~$ cd BT
rajendra@SmartSquareSystems:~/BT$ ls
btcrack-master             libbtbb-master
crackle-master             libffi-3.2.1
glib-2.52.1                PyBT-master
kismet-2013-03-R1b         spectools
kismet-2013-03-R1b.tar.xz  ubertooth-2017-03-R2
libbtbb-2017-03-R2         ubertooth-2017-03-R2.tar.xz
libbtbb-2017-03-R2.tar.gz
rajendra@SmartSquareSystems:~/BT$ cd libbtbb-2017-03-R2/wireshark/plugins/btbb
rajendra@SmartSquareSystems:~/BT/libbtbb-2017-03-R2/wireshark/plugins/btbb$ mkdir build
rajendra@SmartSquareSystems:~/BT/libbtbb-2017-03-R2/wireshark/plugins/btbb$ cd build
rajendra@SmartSquareSystems:~/BT/libbtbb-2017-03-R2/wireshark/plugins/btbb/build$ cmake -DCMAKE_INSTALL_LIBDIR=/usr/lib/x86_64-linux-gnu/wireshark/libwireshark3/plugins ..
-- The C compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Plugin will be installed in: /usr/lib/x86_64-linux-gnu/wireshark/libwireshark3/plugins
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for one of the modules 'glib-2.0>=2.4.0'
-- Found GLib2: glib-2.0 /usr/include/glib-2.0;/usr/lib/x86_64-linux-gnu/glib-2.0/include
-- Looking for include file glib/gregex.h
-- Looking for include file glib/gregex.h - not found
-- Looking for include file glib/gchecksum.h
-- Looking for include file glib/gchecksum.h - not found
-- Configuring done
-- Generating done
-- Build files have been written to: /home/rajendra/BT/libbtbb-2017-03-R2/wireshark/plugins/btbb/build
rajendra@SmartSquareSystems:~/BT/libbtbb-2017-03-R2/wireshark/plugins/btbb/build$ 

### Version information
rajendra@SmartSquareSystems:~/BT/libbtbb-2017-03-R2/wireshark/plugins/btbb/build$ uname -a
Linux SmartSquareSystems 4.10.0-19-generic #21-Ubuntu SMP Thu Apr 6 17:04:57 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

**Ubertooth tools version (ubertooth-rx -V):**
rajendra@SmartSquareSystems:~/BT/libbtbb-2017-03-R2/wireshark/plugins/btbb/build$ ubertooth-rx -V
libubertooth 1.0 (2017-03-R2), libbtbb 1.0 (2017-03-R2)

**libbtbb version:**

**Ubertooth firmware version (ubertooth-util -v):**
rajendra@SmartSquareSystems:~/BT/libbtbb-2017-03-R2/wireshark/plugins/btbb/build$ ubertooth-util -v
Firmware version: 2017-03-R2 (API:1.02)

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
```
Insert any commandline or build output here

Attachments: CMakeError.log CMakeOutput.log

[CMakeError.log.txt](https://github.com/greatscottgadgets/ubertooth/files/962533/CMakeError.log.txt)

[CMakeOutput.log.txt](https://github.com/greatscottgadgets/ubertooth/files/962534/CMakeOutput.log.txt)

Please help.

Thanks,
Raj

dominicgs, 2017-05-04:
Which part of the cmake output do you think is an error?  To me it looks like it configures correctly and you should run `make`

rajpancha1, 2017-05-04:
Dominic:
Thanks for response.

Below is the cmake output with error about not finding the glib/gregex.h and glib/gchecksum.h. 

I did the make and install of Ubertooth tool suites and the thirdparty s/w following the latest build guide.

I don't know what feature did not get built because of that error,

-----------x-------------
cmake -DCMAKE_INSTALL_LIBDIR=/usr/lib/x86_64-linux-gnu/wireshark/libwireshark3/plugins ..
-- The C compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Plugin will be installed in: /usr/lib/x86_64-linux-gnu/wireshark/libwireshark3/plugins
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for one of the modules 'glib-2.0>=2.4.0'
-- Found GLib2: glib-2.0 /usr/include/glib-2.0;/usr/lib/x86_64-linux-gnu/glib-2.0/include
**-- Looking for include file glib/gregex.h
-- Looking for include file glib/gregex.h - not found
-- Looking for include file glib/gchecksum.h
-- Looking for include file glib/gchecksum.h - not found**
-- Configuring done
-- Generating done
-- Build files have been written to: /home/rajendra/BT/libbtbb-2017-03-R2/wireshark/plugins/btbb/build
---------x---------------

Thank you for your kind help.
Raj

dominicgs, 2017-05-04:
Those are not required files, the output below that
```
-- Configuring done
-- Generating done
```
lets us know that cmake succeeded.  You can run 'make' after that.

rajpancha1, 2017-05-04:
Thanks Dominic.

billylees, 2018-12-05:
Those are not required files.
Thanks Dominic.

closed 2020-12-05 by mikeryan