greatscottgadgets/ubertooth #262: Ubertooth & kismet build errors?

dreyes15, 2017-05-01:
I am trying to install the kismet plugin as specified in the third party instructions 
specifically I have run the following:
sudo apt-get install libpcap0.8-dev libcap-dev pkg-config build-essential libnl-3-dev libncurses-dev libpcre3-dev libpcap-dev libcap-dev libnl-genl-3-dev
wget https://kismetwireless.net/code/kismet-2013-03-R1b.tar.xz
tar xf kismet-2013-03-R1b.tar.xz
cd kismet-2013-03-R1b
ln -s ../ubertooth-2017-03-R2/host/kismet/plugin-ubertooth .

I get the following error 

ln: failed to create symbolic link './plugin-ubertooth': File exists

### Version information
Ubuntu 16.04
Ubertooth 2017-03-R2
Kismet-2013-03-R1

dominicgs, 2017-05-02:
Most of those look like permissions errors.  Is this running as root?  Or another user?

Has the ubertooth plugin installed anywhere?  What was the output of the `make install` command for the plugin?

dreyes15, 2017-05-03:
That is the output running as both root and as a regular user, after running: `sudo make plugins-install` I receive the following output 
[image]

Could these errors have originated from the soft link? when I run `ln -s ../ubertooth-2017-03-R2/host/kismet/plugin-ubertooth .`
I get the error that says it failed to create the symbolic link '/plugin-ubertooth': file exists

dreyes15, 2017-05-03:
I have kismet seeing the ubertooth one as a plugin now, I had to change the path of where the source code to kismet was it
[image]

This is what I get when clicking on the plugin, which I think is the correct output?

dominicgs, 2017-05-03:
That looks better.  Try starting some Bluetooth devices nearby to see if you get any output?

dreyes15, 2017-05-03:
There we go I am getting some packets now, thanks!
[image]

dreyes15, 2017-05-03:
Hmm there seems to be another issue It seems after exiting the program the terminal says I have some packets, but upon opening the pcap file it says no packets collected, perhaps now it's a kismet issue?

dreyes15, 2017-05-03:
[image]
Here is a photo of what I output

dominicgs, 2017-05-03:
Bluetooth has a packet scrambler (known as data whitening in the BT spec) that we can't decode when we first receive packets, we need more information about them.  Even when we can decode it, we can't always be sure of the size/modulation of the data section of the packet, so that can often be invalid or left out of the pcap.

It looks like in this case we were never able to unwhiten packet data.

dreyes15, 2017-05-03:
Ok, so in this case it isn't necessarily a Kismet or Ubertooth problem. It is mostly an unwhiten packet data. Alright sounds good then if that's the case then this issue is resolved
Thanks!

dominicgs, 2017-05-04:
I believe it should only produce one output file for Ubertooth captures.

dreyes15, 2017-05-04:
Hmm ok well apart from the packets never being unwhiten it seems like all the issues I was having is fixed.

Bella721, 2018-04-09:
When we use kismet to capture package, why it can't create pacpbtbb files?

mikeryan, 2020-12-05:
Legacy Kismet is no longer supported by this project.

closed 2020-12-05 by mikeryan