greatscottgadgets/ubertooth #264: Can I capture bluetooth classic? and I can't see att packet on wireshark

jbk0310, 2017-05-11:
Hello I'm BK Jang.

I'm using ubuntu 16.04.
I want to capture BLE and BT classic on wireshark so I have tried it as your guide and I can see packet of ble on wireshark but there is not att packet but only adv and scan packet. I used this cmd "ubertooth-btle -f -c /tmp/pipe" How can I see att or the other packet on wireshark?
additional
I want to capture classic packet. I don't know how I can capture packet of BT classic.

tjhasan, 2020-10-04:
I am also not seeing ATT packets. Any help would be appreciated.

mikeryan, 2020-12-05:
For anyone having this problem, you most likely haven't won the race to capture the `CONNECT_REQ` packet. If you don't see that packet in your traces, keep disconnecting and re-establishing the BLE connection and eventually you will capture the connection.

closed 2020-12-05 by mikeryan