greatscottgadgets/ubertooth #266: MacPorts ubertooth-2017-03-R2 install fails

HelterSkelter5, 2017-05-20:
### Steps to reproduce
1. Install MacPorts, Xcode, libbtbb as per install instructions
2. Download and install ubertooth-2017-03-R2 as per install instructions
3. Run cmake .. as per install instructions 

### Expected behaviour
ubertooth-2017-03-R2 build process starts

### Actual behaviour
-- The C compiler identification is AppleClang 8.1.0.8020042
-- The CXX compiler identification is AppleClang 8.1.0.8020042
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Setting version string 2017-03-R2
-- Could NOT find PkgConfig (missing:  PKG_CONFIG_EXECUTABLE) 
-- Found LIBBTBB: /usr/local/lib/libbtbb.dylib  
-- Could NOT find PkgConfig (missing:  PKG_CONFIG_EXECUTABLE) 
-- Could NOT find LIBUSB (missing:  LIBUSB_INCLUDE_DIR) 
-- Building shared library
-- Could NOT find PkgConfig (missing:  PKG_CONFIG_EXECUTABLE) 
-- Could NOT find LIBBLUETOOTH (missing:  LIBBLUETOOTH_LIBRARIES LIBBLUETOOTH_INCLUDE_DIR) 
-- Found PythonInterp: /usr/bin/python (found version "2.7.10") 
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LIBUSB_INCLUDE_DIR (ADVANCED)
   used as include directory in directory /Users/stuart/Downloads/ubertooth-2017-03-R2/host/libubertooth/src
   used as include directory in directory /Users/stuart/Downloads/ubertooth-2017-03-R2/host/libubertooth/src
   used as include directory in directory /Users/stuart/Downloads/ubertooth-2017-03-R2/host/libubertooth/src
   used as include directory in directory /Users/stuart/Downloads/ubertooth-2017-03-R2/host/libubertooth/src
   used as include directory in directory /Users/stuart/Downloads/ubertooth-2017-03-R2/host/libubertooth/src
   used as include directory in directory /Users/stuart/Downloads/ubertooth-2017-03-R2/host/libubertooth/src
   used as include directory in directory /Users/stuart/Downloads/ubertooth-2017-03-R2/host/libubertooth/src
   used as include directory in directory /Users/stuart/Downloads/ubertooth-2017-03-R2/host/ubertooth-tools/src
   used as include directory in directory /Users/stuart/Downloads/ubertooth-2017-03-R2/host/ubertooth-tools/src
   used as include directory in directory /Users/stuart/Downloads/ubertooth-2017-03-R2/host/ubertooth-tools/src
   used as include directory in directory /Users/stuart/Downloads/ubertooth-2017-03-R2/host/ubertooth-tools/src
   used as include directory in directory /Users/stuart/Downloads/ubertooth-2017-03-R2/host/ubertooth-tools/src
   used as include directory in directory /Users/stuart/Downloads/ubertooth-2017-03-R2/host/ubertooth-tools/src
   used as include directory in directory /Users/stuart/Downloads/ubertooth-2017-03-R2/host/ubertooth-tools/src
   used as include directory in directory /Users/stuart/Downloads/ubertooth-2017-03-R2/host/ubertooth-tools/src
   used as include directory in directory /Users/stuart/Downloads/ubertooth-2017-03-R2/host/ubertooth-tools/src
   used as include directory in directory /Users/stuart/Downloads/ubertooth-2017-03-R2/host/ubertooth-tools/src
   used as include directory in directory /Users/stuart/Downloads/ubertooth-2017-03-R2/host/ubertooth-tools/src
   used as include directory in directory /Users/stuart/Downloads/ubertooth-2017-03-R2/host/ubertooth-tools/src
   used as include directory in directory /Users/stuart/Downloads/ubertooth-2017-03-R2/host/ubertooth-tools/src
   used as include directory in directory /Users/stuart/Downloads/ubertooth-2017-03-R2/host/ubertooth-tools/src
   used as include directory in directory /Users/stuart/Downloads/ubertooth-2017-03-R2/host/ubertooth-tools/src
   used as include directory in directory /Users/stuart/Downloads/ubertooth-2017-03-R2/host/ubertooth-tools/src
   used as include directory in directory /Users/stuart/Downloads/ubertooth-2017-03-R2/host/ubertooth-tools/src

-- Configuring incomplete, errors occurred!
See also "/Users/stuart/Downloads/ubertooth-2017-03-R2/host/build/CMakeFiles/CMakeOutput.log".

### Version information
**Operating system**:
macOS 10.12.5

**Ubertooth tools version (ubertooth-rx -V):**
ubertooth-2017-03-R2

**libbtbb version:**
libbtbb-2017-03-R2

**Ubertooth firmware version (ubertooth-util -v):**
N/K

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

dominicgs, 2017-05-20:
It appears to be missing the libUSB header files that it needs.  I'm not familiar with macports, is there a libusb headers package that is separate from the main package?

Is libusb installed?

We don't maintain the config file for the macports system, but I'm happy to help with debugging it so that we can try to correct it.

HelterSkelter5, 2017-05-20:
Yes, libusb is installed.

Not sure if there's another libusb headers package.  Aside from libusb there is also libusb-compat, libusb-devel and libusb-legacy.  Any of these needed?

HelterSkelter5, 2017-05-20:
Just installed the pre-requisites for HackRF One and that's enabled the process to continue...

dominicgs, 2017-05-22:
Thanks for confirming.  It looks like we need to ask the macports people to update their config file

dominicgs, 2017-05-23:
Oh, I just noticed that this wasn't a macports issue as Ubertooth isn't built by macports.  I'll update the prerequisites in the build guide.

Could you let me know which prerequisites for HackRF One you installed?  Where did you find them listed?

HelterSkelter5, 2017-05-23:
> Oh, I just noticed that this wasn't a macports issue as Ubertooth isn't built by macports. I'll update the prerequisites in the build guide.

Smashing!

> Could you let me know which prerequisites for HackRF One you installed? Where did you find them listed?

https://github.com/mossmann/hackrf/wiki/Operating-System-Tips#os-x-105-with-macports

Just followed that one-liner there!

closed 2020-12-05 by mikeryan

Sudococommunity, 2022-10-23:
I am having error no cmakelist found what i do pls give the cmake list if you have