greatscottgadgets/ubertooth #268: CMake error while building libubertooth

Nerade, 2017-06-08:
Compiling ubertooth-tools works fine. Running a binary results in error caused by missing libubertooth.
Installing libubertooth fails because CMake cant find file FindBTBB.cmake module file. 

### Steps to reproduce
1. Download release archive of ubertooth and libbtbb (both version 2017-03-R2)
2. Compile and install libbtbb according to readme
3. Try to install libubertooth according to readme

### Expected behaviour
CMake should create make files for building libbluetooth.

### Actual behaviour
CMake can't find package configuration for BTBB.

### Version information
**Operating system**:
Arch Linux 4.10.13-1

**Ubertooth tools version (ubertooth-rx -V):**
2017-03-R2

**libbtbb version:**
2017-03-R2

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
```
-- Setting version string git-8420a29*
CMake Error at src/CMakeLists.txt:39 (find_package):
  By not providing "FindBTBB.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "BTBB", but
  CMake did not find one.

  Could not find a package configuration file provided by "BTBB" with any of
  the following names:

    BTBBConfig.cmake
    btbb-config.cmake

  Add the installation prefix of "BTBB" to CMAKE_PREFIX_PATH or set
  "BTBB_DIR" to a directory containing one of the above files.  If "BTBB"
  provides a separate development package or SDK, be sure it has been
  installed.
```
Checking manually for FindBTBB.cmake:
`ls ubertooth-2017-03-R2/host/cmake/module/`

Result:
`FindBLUETOOTH.cmake                  FindPythonInterp.cmake  GetGitRevisionDescription.cmake
FindBTBB.cmake                       FindUBERTOOTH.cmake     GetGitRevisionDescription.cmake.in
FindPackageHandleStandardArgs.cmake  FindUSB1.cmake
`

Checking $CMAKE_MODULE_PATH:
`echo "$CMAKE_MODULE_PATH"`

Result:
`/home/nerade/Downloads/ubertooth-2017-03-R2/host/cmake/modules/`

dominicgs, 2017-06-08:
Which directory are you running cmake from? The build instructions expect it to be run from `ubertooth-2017-03-R2/host/build/`, running it from another path could cause this issue.

Nerade, 2017-06-08:
The directory I am in while trying to build is in fact `ubertooth-2017-03-R2/host/libubertooth/build`.
The directory you mentioned is the one I used to build all the ubertooth-tools. running `cmake ../libubertooth/` throws an error because of the previous build of ubertooth-tools.

dominicgs, 2017-06-08:
The build process should build both the tools and the library together.  You should run it as `cmake ..` rather than specifying the directories separately.

dominicgs, 2017-06-08:
Could you paste the output from the build and the error?

Nerade, 2017-06-08:
Depends on which error you mean. Building and installing the host tools throws no error at all. I will deliver other output later when I'm back home.

dominicgs, 2017-06-08:
If you are running the build from the `host/build` directory it will try to build the library as part of that process.  If it's installing the tools then it must either be building the library or giving some output about it.

All of the output from the build process would help.

Nerade, 2017-06-09:
As I managed to find out with your hints the library gets build and installed normally. The problem is the system configuration of arch and possibly all Fedora/RHEL derivates. They are shipped without `/usr/local/lib` beeing configured as valid ld path. Adding this path to `/etc/ld.so.conf` solved the issue for me. Thanks for your help!

closed 2017-06-09 by Nerade