greatscottgadgets/ubertooth #270: Ubertooth-btle freezes on CONNECT_REQ

Nerade, 2017-06-09:
### Steps to reproduce
1.  Run ubertooth-btle -f -c /tmp/pipe
2.  Configure wireshark to use pipe as input
3.  Initiate connection attempt with 2 btle devices
4.  Repeat until `CONNECT_REQ` is found

### Expected behaviour
Ubertooth sniffing should follow the connection and receive data from this connection.

### Actual behaviour
ubertooth-btle seems to freeze after the `CONNECT_REQ` package; no more traffic is logged. Only restarting the `ubertooth-btle` command resolves this state.

### Version information
**Operating system**:
Arch Linux 4.10.13-1

**Ubertooth tools version (ubertooth-rx -V):**
libubertooth 1.0 (git-8420a29*), libbtbb 1.0 (git-8420a29*)

**Ubertooth firmware version (ubertooth-util -v):**
2017-03-R2 (API:1.02)

** Wireshark version:**
version 2.2.6

[connect_req.zip](https://github.com/greatscottgadgets/ubertooth/files/1063921/connect_req.zip)

GimliSonOfGloin, 2017-08-16:
I see this occur at times as well. Only difference for me is that I am storing the pcap into a file. At times it can go 10 hours without issue, other times it goes fifteen minutes. All firmware/tools are the same. I have the latest version of Kali as of the date of this post.

Anyone find a fix to this issue?

nemhods, 2017-12-19:
Could this be related to the bug mentioned in https://github.com/greatscottgadgets/ubertooth/blob/master/host/README.btle.md?

> The code always assumes all data channels are in use. I have never observed otherwise, but it's within spec to use fewer.

I am seeing freezes as well, and the CONNECT_REQUEST in question indicates a channel map of `0x0080ffff1f` - meaning the connection will NOT use all data channels. ubertooth-btle would therefore make incorrect assumptions about the channel hopping scheme, and will therefore not be able to follow subsequent packets.

mikeryan, 2017-12-19:
If you're seeing this, you can try the experimental firmware in the le_phy branch that does have support for partial channel maps. I would appreciate any and all testing!

nemhods, 2017-12-20:
The firmware from the le_phy branch resolves this issue for me. Looking forward to seeing this in the next main release too! Thank you for your work.

ceesb, 2018-02-21:
Hi @mikeryan,

I really like the ubertooth project, but for some reason, dumps made with the le_phy branch show corrupted packets:

[image]

Console output of "ubertooth-btle" is giving errors indicating not all packet data ends up on the host:

```
systime=1519190571 freq=2402 addr=8e89bed6 delta_t=13.907 ms rssi=-92
40 25 19 05 00 10 40 16 02 01 06 11 07 00 00 0a ed 39 85 83 ed da 8d 43 6f bc 33 16 3e 0c 13 dc c0 ad a9 f1 e3 af 9c ec b2 ee 
Advertising / AA 8e89bed6 (valid)/ 37 bytes
    Channel Index: 37
    Type:  ADV_IND
    AdvA:  16:40:10:00:05:19 (random)
    AdvData: 02 01 06 11 07 00 00 0a ed 39 85 83 ed da 8d 43 6f bc 33 16 3e 0c 13 dc c0 ad a9 f1 e3 af 9c
        Type 01 (Flags)
           00000110
               LE General Discoverable Mode
               BR/EDR Not Supported

        Type 07 (128-bit Service UUIDs)
           3e1633bc-6f43-8dda-ed83-8539ed0a0000
Error: attempt to read past end of buffer (22 + 12 > 31)

    Data:  19 05 00 10 40 16 02 01 06 11 07 00 00 0a ed 39 85 83 ed da 8d 43 6f bc 33 16 3e 0c 13 dc c0 ad a9 f1 e3 af 9c
    CRC:   ec b2 ee
```

With the master branch it looks fine, but then I cannot follow CONN_REQs:

[image]

I tried with git-84e3d01, both host tools and firmware.

Any clues on what I'm doing wrong with the le_phy branch?

nficano, 2018-05-29:
I'm also having this issue, I tried upgrading to the firmware in le_phy and master and each time after its flashed, only the green led is illuminated and I need to use a jumper to flash it back to the release firmware. Any idea what I could be doing wrong here? (Platform Ubuntu Server 18.01 (not running in a vm))

mikeryan, 2018-06-27:
This issue is fixed in the latest pre-release 2018-06-R1.

closed 2018-06-27 by mikeryan