greatscottgadgets/ubertooth #271: Ubertooth One devices do not show in "lsusb", only green LEDs are on

zcsizmadia, 2017-06-12:
We have couple of Ubertooth One devices (r161-p11) which do not show up in the Linux OS with lsusb. Two green LEDs turn on (LED RST and LED 1V8). They were working before, but they just got stopped working after a power cycle. 

Tried to program them with a LPC ISP using the PROGRAM header (bootloader + bluetooth_rxtx). Programming seems to be successfull, but it looks the device itself fails to boot up fully.

Any thought which component could have failed?

Thanks in advance!

### Expected behaviour
Device should show in "lsusb"

### Actual behaviour
Device is not shown in "lsusb"

### Version information
**Operating system**:
Ubuntu 16.04.2

**Ubertooth tools version (ubertooth-rx -V):**
libubertooth 1.0 (git-3cb9c41)

**libbtbb version:**
libbtbb 1.0 (git-3cb9c41)

**Ubertooth firmware version (ubertooth-util -v):**
2017-03-R2-V1 (API:1.02)

dominicgs, 2017-06-16:
If you connect the pins to enter DFU mode as you plug in the Ubertooth (https://pbs.twimg.com/profile_banners/19587312/1354079866/1500x500) do you see anything happen?

gudaguang, 2020-05-26:
I also meet the same problem. Ubertooth one worked normally yesterday, but now any platform OS(I tried Ubuntu, Windows, Mac) can not detect usb device for Ubertooth.

any advice on that? Appreciate for it.

gudaguang, 2020-05-26:
BTW, if connect the pins to enter DFU mode, LED flashing pattern seems changed, but still no usb device can be enumerated.

mikeryan, 2020-12-05:
If anyone else has this issue, please open a new issue and we will address it.

closed 2020-12-05 by mikeryan