greatscottgadgets/ubertooth #273: Ubertooth not being recognized at all

fencingparry4, 2017-06-16:
I'm using a chromebook running ubuntu (not in a VM) and I have been unable to get the computer to see the ubertooth. No commands show the device running and no ubertooth commands are able to find the device. For example, using `lsusb` does not show any devices. Did I do something wrong in setup? (I did all the commands in the build guide from the root directory (/)). There is no way for me to grab the firmware of the device and even touching the pins together as it says in `info ubertooth-dfu`.

It seems like the problem isn't from the ubertooth software itself - its almost as if the board is ejecting itself as soon as its inserted and theres no way to read or write to it.

There are only 2 green lights running - does that mean something?

dominicgs, 2017-06-16:
If you run `dmesg` after inserting the Ubertooth, do you get any output related to USB devices being connected?

fencingparry4, 2017-07-06:
Sorry for the EXTREMELY late response (i went on vacation), I ran the command `dmesg | tail -20` and there were no lines relating to a USB device being inserted

dominicgs, 2017-07-06:
Are you able to plug it in to any other system?

Where did you buy your Ubertooth?

fencingparry4, 2017-07-10:
When I plugged the ubertooth into a raspberry pi, no USB device was recognized in `dmesg | tail -20` - we bought the ubertooth from NooElec off of Amazon

dominicgs, 2017-07-10:
Could you try connecting the two pins in this image as you plug the Ubertooth in?
https://pbs.twimg.com/profile_banners/19587312/1354079866/1500x500

It should put it in to firmware update mode and the LEDs should flash

fencingparry4, 2017-07-11:
I tried doing that and nothing happened - good idea though

dominicgs, 2017-07-11:
It sounds like a hardware problem.  Could you contact NooElec and say that I have helped you diagnose it as a defective unit.  They should replace it for you.

closed 2020-12-05 by mikeryan