greatscottgadgets/ubertooth #275: Including libusb.h

rkrejci, 2017-06-23:
Hi,
we would like to use libubertooth in our tool. However, when including `ubertooth.h`, it gets into `ubertooth_control.h` which includes `libusb.h`. Unfortunately, `libusb.h` is not directly in any system include directory, but it is installed **always** into `libusb-1.0` subdirectory. libubertooth goes through this thanks to libusb's pkgconfig file which provides correct path.

In our case, we would need to check for libusb in our cmake, but we actually does not use libusb directly (it is hidden inside libubertooth). It seems to me quite uncommon to check for it just because of indirect dependency. The simplest solution to me is to include `libusb-1.0/libusb.h` in `ubertooth_control.h`. Another option is to have pkgconfig file for libubertooth (cmake module to find ubertooth actually use it, but I didn't find the .pc file in the repository) providing also the include path for `libusb.h`.

Is the first option acceptable for you or do you have some other opinion?

dominicgs, 2017-06-23:
My first thought is that we shouldn't require applications that consume libUbertooth to know about libUSB.  The point of libUbertooth is to be an abstraction layer, so we should be able to hide libUSB within it and not require you to have the header (obviously you'd still need it for building libUbertooth)

rkrejci, 2017-06-29:
Sorry for the delay, I was out for a couple of days. 

That should be the best solution, but it seems as a lot of work / big change to me , doesn't it? If I understand it correctly, your idea is to hide `libusb_device_handle` behind some libubertooth structure, right? Personally I'm fine with such a change, but all the tools using the affected functions will have to be changed.

dominicgs, 2017-06-29:
I tried this last Friday night and you're right, it's much bigger than I originally thought.  I even started down the road of refactoring the entire library, which is a task that I don't have time for right now.

As a work around until I can get to it, could you use our cmake module to find libusb 1.0 the way we do?  Using this: https://github.com/greatscottgadgets/ubertooth/blob/master/host/cmake/modules/FindUSB1.cmake

rkrejci, 2017-06-30:
yes, we already do to get the include option for the compiler. This is not something critical, it would be just nice to have it right and don't bother users with handling library used somewhere in the lower layer..

commit 720c87dd (2020-12-12): host: include libusb-1.0/ in include path for libusb.h

Fixes #275
closed 2020-12-12 by mikeryan