greatscottgadgets/ubertooth #276: ubertooth-scan -s giving no output

sgript, 2017-06-24:
### Steps to reproduce
I simply ran `ubertooth-scan -s`.

### Expected behaviour
I should be seeing the BLE devices.

### Actual behaviour
However, I just get the log below, with "HCI scan" with nothing appearing even after a prolonger period of time. I'm scanning for BLE-enabled smartwatches (Pebbles).

### Version information
**Operating system**:
Ubuntu 16.10 (on VMWare Fusion 8 (Version 8.5.7)

**Ubertooth tools version (ubertooth-rx -V):**
libubertooth 1.0 (2017-03-R2)

**libbtbb version:**
libbtbb 1.0 (git-1b6084d)

**Ubertooth firmware version (ubertooth-util -v):**
2017-03-R2 (API:1.02)

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

Running crackle in attempts to try find LTK keys from devices.

### Output
```
osboxes@osboxes:~/lab/ble/spectools$ ubertooth-scan -s
HCI scan

```

dominicgs, 2017-06-24:
Ubertooth-scan does not handle BLE traffic, only BR. It expects a timeout
arguments, otherwise it runs until you ctrl+c it.

sgript, 2017-06-25:
Hi @dominicgs! Thanks for your quick response.

I am having an issue after this initial issue - I was able to get output, but not for my HCI scan. I've tried on two different VMs and have had different results. 

The issue is that on one of my VMs I can actually get the BT Address for my smartwatch as so:
[image]

Whereas another (this one) does not produce such output, would you happen to know what could be causing this?

dominicgs, 2017-07-06:
It may be related to permissions to use the Bluetooth controller in the system.  The HCI scan part of `ubertooth-scan` uses the built in Bluetooth controller rather than Ubertooth.  It may be that you need to set up the VM to take control of the BT controller from the host.

(Sorry for the slow response this time!)

closed 2018-03-18 by sgript