greatscottgadgets/ubertooth #277: Change Ubertooth BT Address

eli80, 2017-07-13:
Hello!

I'm working with Ubertooh Release 2017-03-R2.

I'd like to change device's MAC address, but I can't find the way.

Programms like spooftooph and btaddr seams not working because of a 'manufacturer unknown' error.

Is there any way to achieve the goal?

Thank You

dominicgs, 2017-07-13:
Ubertooth is not a Bluetooth device, it doesn't have a MAC address.  `btaddr` will work with many common USB Bluetooth dongles.

eli80, 2017-07-13:
My apologies, I'm using also the HW device ubertooth 1. 
After your answet I guess i should ask the question to the HW producer (if I find a contact)
Thank you anyway

dominicgs, 2017-07-13:
The Ubertooth One device is not a Bluetooth dongle.

eli80, 2017-07-14:
Sorry for my incompetence, I have no experience with this kind of things.
I know Ubertooth One is a developer platform, but actually it has a MAC address, I can see it in Kali.
Since it's not a Bluetooth dongle btaddr is not working and I can understand why, but I can't understand if there's another way to change it.
I though it's address was defined in the firmware, so that's why I posted my question here.

dominicgs, 2017-07-18:
It's not incompetence, there's definitely a steep learning curve with these tools.

Ubertooth does not have a MAC address.  When you say that you can see it in Kali, what tool are you using?

eli80, 2017-07-24:
I'm starting to understand something.. 
The MAC address I saw was the one associated to the computer BT module.. actually I though I was using Ubertooth from Kali, but in reality I was using the native computer BT module.
I'm using Kali on a Virtual Machine and this was probably  causing some interferences  between the two modules.
What I got is that current Ubertooth, without additional customer SW, can be mainly used for sniffing BLE traffic. 
Is it correct?
Instead, its capability for Classic BT is not clear. It can't follow the frequency  hopping in this case, right?
Am I missing something?

dominicgs, 2017-08-11:
You are correct.  Ubertooth is good for BLE and for finding unknown devices using classic BT. However, we have code that will attempt to follow the hopping pattern, but it's rather unreliable.

closed 2018-06-27 by mikeryan