greatscottgadgets/ubertooth #278: Errors building plugin-ubertooth

aaronjeline, 2017-07-13:
### Steps to reproduce
1. Kismet Source installed to /usr/src/kismet
2. In host/kismet/plugin-ubertooth
3. Attempt 'sudo make install'

### Expected behaviour
Plugin compiles

### Actual behaviour
Compilation Fails:
`g++ -I./msgpack-1.4.1/include   -pthread  -fopenmp -Wall -g -I. -g -O2 -I/usr/src/kismet -I/usr/include/libusb-1.0 -I/usr/local/include/libusb-1.0 -I/usr/include -g -fPIC -c packetsource_ubertooth.cc -o packetsource_ubertooth.o 
packetsource_ubertooth.cc:32:26: fatal error: packetsource.h: No such file or directory
 #include <packetsource.h>
`

### Version information
**Operating system**: Fedora 25 64-bit

**Ubertooth tools version (ubertooth-rx -V):** 2017-03-R2

**libbtbb version:** 1,0

**Ubertooth firmware version (ubertooth-util -v):** 2017-03-R2 (API: 1.02)

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.
Kismet: 2017 (latest from git)

### Output
```
g++ -I./msgpack-1.4.1/include   -pthread  -fopenmp -Wall -g -I. -g -O2 -I/usr/src/kismet -I/usr/include/libusb-1.0 -I/usr/local/include/libusb-1.0 -I/usr/include -g -fPIC -c packetsource_ubertooth.cc -o packetsource_ubertooth.o 
packetsource_ubertooth.cc:32:26: fatal error: packetsource.h: No such file or directory
 #include <packetsource.h>

```

aaronjeline, 2017-07-13:
(I can't read, sorry for wasting your time)

closed 2017-07-13 by aaronjeline

dominicgs, 2017-07-13:
Kismet has moved on a lot recently and we haven't had chance to get the plugins up to date yet, but I hope to get time for them later in the year.

aaronjeline, 2017-07-13:
Yeah just saw that in a different page on the wiki. The Getting Started page doesn't specify which version you have to use.