greatscottgadgets/ubertooth #279: Displaty error with specscan-ui

aaronjeline, 2017-07-13:
### Steps to reproduce
1. Run ''' sudo ubertooth-specan-ui

### Expected behaviour
UI should run

### Actual behaviour
White window appears, exception thrown

### Version information
**Operating system**: Fedora 25, 64-bit

**Ubertooth tools version (ubertooth-rx -V):** 2017-03-R2

**libbtbb version:** 1.0

**Ubertooth firmware version (ubertooth-util -v):** 2017-03-R2

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
```
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib64/python2.7/threading.py", line 804, in __bootstrap_inner
    self.run()
  File "/usr/local/bin/ubertooth-specan-ui", line 50, in run
    for frequency_axis, rssi_values in frame_source:
  File "/usr/lib/python2.7/site-packages/specan/Ubertooth.py", line 46, in specan
    self.proc = subprocess.Popen(args, stdout=subprocess.PIPE, stderr=subprocess.PIPE)
  File "/usr/lib64/python2.7/subprocess.py", line 390, in __init__
    errread, errwrite)
  File "/usr/lib64/python2.7/subprocess.py", line 1024, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory

X Error: BadAccess (attempt to access private resource denied) 10
  Extension:    130 (MIT-SHM)
  Minor opcode: 1 (X_ShmAttach)
  Resource id:  0x10b
X Error: BadShmSeg (invalid shared segment parameter) 128
  Extension:    130 (MIT-SHM)
  Minor opcode: 3 (X_ShmPutImage)
  Resource id:  0x120000b
X Error: BadShmSeg (invalid shared segment parameter) 128
  Extension:    130 (MIT-SHM)
  Minor opcode: 3 (X_ShmPutImage)
  Resource id:  0x120000b
X Error: BadShmSeg (invalid shared segment parameter) 128
  Extension:    130 (MIT-SHM)
  Minor opcode: 3 (X_ShmPutImage)
  Resource id:  0x120000b

```

aaronjeline, 2017-07-13:
Partially solved it.
The place the binaries are stored (/usr/local/bin) are not on the default path for root on fedora.
That solves the file open error.
However, I still get the BadAccess and BadShmSeg, and a white window.

dominicgs, 2017-07-14:
I believe those errors come from running as root too.  If you install the udev rule from the repository, then you can use the Ubertooth without being root.

Alternatively, you may be able to run `sudo QT_X11_NO_MITSHM=1 ubertooth-specan-ui`

dragonxtek, 2017-07-20:
I have this same error. But when I run _sudo python setup.py install_ and I run _sudo ubertooth-specan-ui_ I get the following message:
**Traceback (most recent call last):
  File "/usr/local/bin/ubertooth-specan-ui", line 32, in <module>
    from specan import Ubertooth
  File "/usr/local/lib/python2.7/dist-packages/specan/Ubertooth.py", line 24, in <module>
    import usb.core
ImportError: No module named usb.core**
I run:
**sudo pip install libusb 
sudo pip install pyusb   
sudo apt-get install libusb-dev** 

but I can't fix it. Any idea?

dominicgs, 2017-07-20:
@dragonxtek The error message that you list does not match the error on this issue.  I believe you are experiencing a different issue.  Please follow the build guide here: https://github.com/greatscottgadgets/ubertooth/wiki/Build-Guide

dragonxtek, 2017-07-20:
I have the same issue, when I run ubertooth-specan-ui in the project folder

dominicgs, 2017-07-20:
Did you try the fixes that I suggested in my previous comment - installing the udev rule, or running as `sudo QT_X11_NO_MITSHM=1 ubertooth-specan-ui` ?

dragonxtek, 2017-07-22:
Yes, runing `sudo QT_X11_NO_MITSHM=1 ubertooth-specan-ui` returns the same issue than #280

closed 2020-12-05 by mikeryan