greatscottgadgets/ubertooth #280: TypeError: __init__() takes exactly 2 arguments (1 given)

dragonxtek, 2017-07-20:
### Steps to reproduce
1.- sudo pip install pyusb libusb  
2.- sudo apt-get install libusb-dev python-usb  
3.- sudo python setup.py install
4.- cd ~
5.- sudo ubertooth-specan-ui

### Expected behaviour
Execute ubertooth-specan-ui correctly

### Actual behaviour
I got the following error:
**Traceback (most recent call last):
  File "/usr/local/bin/ubertooth-specan-ui", line 377, in <module>
    window = Window(ubertooth_device=ubertooth_device)
  File "/usr/local/bin/ubertooth-specan-ui", line 285, in __init__
    self._device = self._open_device()
  File "/usr/local/bin/ubertooth-specan-ui", line 300, in _open_device
    return Ubertooth.Ubertooth()
TypeError: __init__() takes exactly 2 arguments (1 given)**

### Version information
**Operating system**: Ubuntu 16.04.2 64 bits

**Ubertooth version**: ubertooth (2017-03-R2)

dominicgs, 2017-07-20:
Did you also install `python-numpy python-pyside python-qt4`?

I don't recommend using pip to install dependencies like this: `pip install pyusb libusb`, especially as you `apt-get install python-usb` right afterwards.  You may have two copies of pyUSB installed to different locations.

dragonxtek, 2017-07-20:
Yes, I have them installed

dominicgs, 2017-07-20:
Did you check to see if you have two copies of pyUSB installed?

dragonxtek, 2017-07-22:
I removed pip pyUSB installation and it returned the same error.
How can I check if I have two copies or more in my system?

closed 2020-12-05 by mikeryan