greatscottgadgets/ubertooth #281: Unable to sniff the pairing request and response between Mobile and BLE device

ByteAlone, 2017-07-20:
Hi.
I'm using Ubertooth one with Wireshark in order to sniff BLE traffic between my Mobile app and a BLE device.
The sniffer works well; I captured both advertising and data packets.
Also all the CONNECT_REQ, LL_FEATURE_REQ, LL_FEATURE_RSP and all the various Reading and Writing request and response are captured.

The only messages that I can't see in .pcap file are the Pairing Request and Response., with all the SMP BLE packets.
Am I missing something?
Why are these packets missing?
What can be the cause?

I tried with different BLE devices, and the result is the same.

Many thanks.

### Version information
**Ubertooth tools version (ubertooth-rx -V):**libubertooth 1.0 (2017-03-R2), libbtbb 1.0 (2017-03-R2)
**Ubertooth firmware version (ubertooth-util -v):** Firmware version: 2017-03-R2 (API:1.02)

hunterzju, 2017-07-24:
maybe it is not working in pair mode

ByteAlone, 2017-07-24:
I've tested different devices and finally I realized that none of them use the bonding procedure (the pairing never starts). The app simply connect to the device and access its characterisitc without pairing first.

ByteAlone, 2017-07-24:
I've tested different devices and finally I realized that none of them use the bonding procedure (the pairing never starts). The app simply connect to the device and access its characterisitc without pairing first.
Many thanks for the answer.

closed 2017-07-24 by ByteAlone