greatscottgadgets/ubertooth #282: Firmware update

Kay887, 2017-07-22:
### Steps to reproduce
1.
2.
3.

### Expected behaviour
When I try to update the firmware on the ubertooth-one, it gives me a long string of "control message unsupported" message. I've set up the requirements for ubertooth-one 2017 version. Ideally, it should be able to recognize the ubertooth-one to flash the device

### Actual behaviour
When I type in ubertooth-dfu -d bluetooth_rxtx.dfu -r, i get a long string of control message unsupported. 

### Version information
I am running kali 2017 installed on my hard disk

**Ubertooth tools version (ubertooth-rx -V):** control message -9

**libbtbb version:**

**Ubertooth firmware version (ubertooth-util -v):**

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

dominicgs, 2017-08-11:
Did you manage to update your Ubertooth?  I think you need to put it in to DFU mode manually by connecting pins 1 and 3 on the expansion header.

Kay887, 2017-08-18:
@dominicgs I jumpered the ubertooth with the help of a pin and I was able to update the firmware. 

Thanks for the help.

Wanted to ask you for ubertooth packet injection, I was not able to find any tutorial or article. Could you please guide me towards it. I am relatively new to bluetooth?

Thanks

dominicgs, 2017-08-18:
I don't have any packet injection code for the Ubertooth.

Kay887, 2017-08-18:
Thanks for the reply. I was just curious to understand what will the ubertooth-tx command do? Won't I be able to inject some data packets into a BLE communication? 

I also tried to use the ubertooth in the faux slave mode. As soon I enter a bogus mac address and perform hcitool lescan, for some reasons. the mac address does not reflect in the list?

Thanks

closed 2020-12-05 by mikeryan