greatscottgadgets/ubertooth #283: undefined reference to `btbb_piconet_clear_channel_seen'

dragonxtek, 2017-07-22:
### Steps to reproduce
1. sudo apt-get install libbtbb-dev 
2. install libbtbb from github
3. cd ubertooth/host/
4. mkdir build
5. cd build
6. cmake ..
7. make

### Expected behaviour
compile files

### Actual behaviour

undefined reference to `btbb_piconet_clear_channel_seen' doesn't enable compile the code

### Version information
**Operating system**: Ubuntu 16.04.2 LTS 64 bits

**libbtbb version:** 2017-03-R2

### Output
```
[  3%] Building C object libubertooth/src/CMakeFiles/ubertooth.dir/ubertooth.c.o
[  6%] Building C object libubertooth/src/CMakeFiles/ubertooth.dir/ubertooth_callback.c.o
[  9%] Building C object libubertooth/src/CMakeFiles/ubertooth.dir/ubertooth_control.c.o
[ 12%] Building C object libubertooth/src/CMakeFiles/ubertooth.dir/ubertooth_fifo.c.o
[ 15%] Linking C shared library libubertooth.so
[ 15%] Built target ubertooth
Scanning dependencies of target ubertooth-debug
[ 18%] Building C object ubertooth-tools/src/CMakeFiles/ubertooth-debug.dir/ubertooth-debug.c.o
[ 21%] Building C object ubertooth-tools/src/CMakeFiles/ubertooth-debug.dir/cc2400.c.o
[ 25%] Building C object ubertooth-tools/src/CMakeFiles/ubertooth-debug.dir/arglist.c.o
[ 28%] Linking C executable ubertooth-debug
../../libubertooth/src/libubertooth.so.1.0: undefined reference to `btbb_piconet_clear_channel_seen'
collect2: error: ld returned 1 exit status
ubertooth-tools/src/CMakeFiles/ubertooth-debug.dir/build.make:150: recipe for target 'ubertooth-tools/src/ubertooth-debug' failed
make[2]: *** [ubertooth-tools/src/ubertooth-debug] Error 1
CMakeFiles/Makefile2:230: recipe for target 'ubertooth-tools/src/CMakeFiles/ubertooth-debug.dir/all' failed
make[1]: *** [ubertooth-tools/src/CMakeFiles/ubertooth-debug.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

```

jjhein, 2017-12-07:
I had the same issue, but it was because I had the libbtbb* packages installed.  Try the following:

sudo apt-get remove libbtbb-dev libbtbb0 libbtbb0-dbg python-libbtbb-pcapdump

Remove your build directory, create a new one and rerun cmake.

You should be good to go.

dominicgs, 2017-12-07:
@jjhein is correct you need matching versions of libbtbb and the Ubertooth tools.  If you want to install from a package manager, install both from the package manager, but if you want to build from source, you need to build both from source.

closed 2017-12-07 by dominicgs

dragonxtek, 2017-12-07:
Thanks, now I updated the process to install and how to deal with any problems that may arise in my blog.
 [https://installfights.blogspot.cl/2016/07/install-and-configure-ubertooth-one.html](https://installfights.blogspot.cl/2016/07/install-and-configure-ubertooth-one.html)