greatscottgadgets/ubertooth #286: Unable to sniff btle traffic from Apple Watch & Smart Watch 3

pentagramz, 2017-09-03:
### Steps to reproduce
1. Unpair the watch and phone

2. Start sniff 

3. Start pairing 

4. Capture BLE traffic 

5.Once watch restore done...wait for some time 

6. Then unpair the devices 

7. Analyse the captured traffic using crackle

### Expected behaviour
I am not able to see conne_req..please help unable to crack key

### Actual behaviour
Tell us what happens instead

Instead it shows unable to crack Please see screenshots

[image]
[image]
[image]

### Version information
**Operating system**:
Host machine Windows 7 64 bit, running VMware player 
Ubertooth one connected to VMware , and using inbuilt bluetooth adapter
Running commands in lubuntu machine running in VMWare player

**Ubertooth tools version (ubertooth-rx -V):**

libubertooth 1.0 (2017-03-R2), libbtbb 1.0 (2017-03-R2)

**libbtbb version:** libbtbb 1.0 (2017-03-R2)

**Ubertooth firmware version (ubertooth-util -v):** Firmware version: 2017-03-R2 (API:1.02)

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

pentagramz, 2017-09-03:
Command used ubertooth-btle -f -A 38 -t4C:32:75:F0:84:1A -r 15.pcapng

channels tried changing to 37, 38  , 39

pentagramz, 2017-09-03:
Command used ubertooth-btle -f -A 38 -tBD ADDR -r 15.pcapng

channels tried changing to 37, 38 , 39

nemhods, 2017-12-19:
At least for one of the errors you see: It seems like your BLE device uses Bluetooth 4.2's _Secure Connections_ feature. There is currently no known flaw in the involved protocols that would allow you to crack the encryption key. The device is resistant to cracking for now.

closed 2018-06-27 by mikeryan