greatscottgadgets/ubertooth #287: Ubertooth BT Classic issue

vinu123456789, 2017-09-05:
Hii I'm Vinod

Iam trying to capture classic packets while sending audio from mobile to my ubuntu machine and steps i followed - 
1.started ubertooth sniffing by using cmd - "sudo ubertooth-rx -q classic_5.pcap".
2.started scanning and pairing in my ubuntu machine .
3.i connected to my mobile also.
4.i sent audio.

output..
iam not able to see any packet related my mobile or ubuntu connection it is just showing only one lap.that is also i dont know which devices lap.

Iam using 
Ubuntu version 	       -  17.04
ubertooth one hostcode -  2017-03-R2
libbtbbb	       -  2017-03-R2
(libubertooth 1.0 (2017-03-R2), libbtbb 1.0 (2017-03-R2))
(Firmware version: 2017-03-R2 (API:1.02))
wireshark		- 2.2.6

output of cmd
[image]

in this u can see lap address "57a8f1" but i dont which devices lap it is.

so please help me in this issue.

dominicgs, 2017-09-07:
Do you know the address of your two devices?  I would expect that LAP to belong to one of them.

The LAP shown will be the LAP of the master device, that's the lower three bytes of the address of one device.

Even if that's your device, Ubertooth is unable to follow a connection and recover 100% of the packets.  Those that are recovered will likely be encrypted.

vinu123456789, 2017-09-09:
Hello Domi,

Thanks a lot for your quick reply !!!.

I am aware of two devices. Here one device is my phone & other one is ubuntu linux on my desktop.

I able to see only advertise packets not able to capture the data packets. 
 
Please let me know if ubertooth supports to capture the Data packets. If yes please let me know how to proceed that.

WitcherDev, 2018-04-09:
I recommend to read wiki, there is ubertooth-rx command for it

fazega, 2018-05-10:
I understand all of this.
But why is there so few information in each packet ? For instance, an inquiry and its answer does not need synchronisation, right ? Since it's broadcast ?
Simply said : why can't we have an interface like : this packet is coming from LAP1 to LAP2, and it's inquiry/data exchange.

dominicgs, 2018-05-12:
Bluetooth devices do not listen to all channels simultaneously.

Bluetooth chips always know what mode they're in, they listen only for packets that they are interested in and throw the rest away.  We have to listen for all packets and try to determine what mode the sender and receiver are in.

Bluetooth discovery is designed to spam small packets across many channels so that hopefully a receiving device sees one and responds.  We don't know which channel a device is listening on at any given time, so we cannot know which channel a device may respond on.

fazega, 2018-05-12:
Ok I got it !
Thank you !

closed 2020-12-05 by mikeryan