greatscottgadgets/ubertooth #288: Ubertooth-Kismet No Longer Builds

decidedlygray, 2017-09-08:
### Steps to reproduce
1. Follow all steps in `host/kismet/plugin-ubertooth/README`, cloned kismet to `/usr/src/` to avoid issues
2. make

### Expected behaviour
Expected the usbertooth-kismet plugin to compile

### Actual behaviour
Does not compile because of a change to Kismet (packetsource.h no longer exists in kismet/master).

### Version information
**Operating system**: Kali

**Ubertooth tools version (ubertooth-rx -V):** libubertooth 1.0 (2017-03-R2)

**libbtbb version:** libbtbb 1.0 (2017-03-R2)

**Ubertooth firmware version (ubertooth-util -v):** Firmware version: 2017-03-R2 (API:1.02)

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

Kismet, latest clone of https://github.com/kismetwireless/kismet.
```
# git rev-parse HEAD
339fa8ae568e092cd4c5ac02c2417946065527e0
```

### Output
```
g++ -pthread  -fopenmp -I./msgpack-1.4.1/include  -Wall -g -I. -g -O2 -I/usr/src/kismet -I/usr/include/libusb-1.0 -I/usr/local/include/libusb-1.0 -I/usr/include -g -fPIC -c packetsource_ubertooth.cc -o packetsource_ubertooth.o 
packetsource_ubertooth.cc:32:10: fatal error: packetsource.h: No such file or directory
 #include <packetsource.h>
          ^~~~~~~~~~~~~~~~
compilation terminated.
Makefile:54: recipe for target 'packetsource_ubertooth.o' failed
make: *** [packetsource_ubertooth.o] Error 1
```

Going to where kismet was cloned (`/usr/src/kismet/`) and looking for the file, I couldn't find it.

So I went back to the Kismet repo and searched the git history a little bit, until I found this:
https://github.com/kismetwireless/kismet/commit/b98c9585d82709dfd6d966fb969cd24d539bcda7

TL;DR - About 5 months ago it looks like Kismet was heavily refactored, and `packetsource.h` no longer exists in the main build. It is being kept in a `legacy-code` folder for archive's sake, but `/usr/src/kismet/legacy_code` can't be passed as KIS_SRC_DIR because `legacy_code` is not a full source archive, it is just a dumping ground for deprecated files.

decidedlygray, 2017-09-08:
I went deeper down the rabbit hole, attempting to git reset --hard <commit> to older commits. Even a year ago doesn't build. I also tried removing the packetsource.h dependency and swapping in what looks like its replacement kis_datasource.h, and renaming some of the classes. But there are other things that need to be fixed once this happens.

dominicgs, 2017-09-11:
Kismet is a rapidly moving target right now.  The regular plugin builds against the last stable release.  The phy-neutral plugin built against git head until the large refactor, but I'm not in a position to keep up until the development of kismet stabalises.

dominicgs, 2017-10-02:
I don't use the Kismet plugins myself, but I'm fairly sure that the last stable Kismet release (2016 ?) works with the plugin in the [plugin-ubertooth](https://github.com/greatscottgadgets/ubertooth/tree/master/host/kismet/plugin-ubertooth) directory.

dominicgs, 2017-10-02:
Let me know if that's not the case.

IanLowe, 2017-10-02:
Thanks for the quick response - I'll check and get back to you :)

IanLowe, 2017-10-12:
Hi, so, I didn't get a 2016 era Kismet to work (easily), but dropped back to Kismet 2013-03-R1b - that works as expected with the plugins in 2017-03-R2.

EMCP, 2018-01-02:
Ran through instructions, rebooted, confirmed tool is working on command line with `ubertooth-rx -z`

but when in Kismet seeing no sources to log, please input a source

marceldeb, 2018-03-16:
Does the plugin even works on the latest version of Kismet @IanLowe 
If so, could you be so kind to tell exactly how you were able to build it?
I also downloaded the old 2013-03-R1 release and pointed the KIS_SRC_DIR to this folder. However it showed me an error it could not find the makefile.inc

mikeryan, 2020-12-05:
Legacy Kismet is no longer supported by this project.

closed 2020-12-05 by mikeryan

atroche, 2021-03-25:
@mikeryan do you know which Git SHA of the Kismet source will allow `plugin-ubertooth-phyneutral` to build? Being on latest for both repos produces an error about a missing include (`packetsource.h`).

Or is the Kismet Ubertooth plugin not supported at all anymore?

Cheers!

ZeroChaos-, 2021-03-25:
the kismet plugin is not supported at all anymore.  the current version of kismet has support for using ubertooth to sniff btle without the plugin, although no support for sniffing bluetooth classic yet.

sagfed, 2021-12-04:
> the kismet plugin is not supported at all anymore. the current version of kismet has support for using ubertooth to sniff btle without the plugin, although no support for sniffing bluetooth classic yet.

When it is stated that the current version of Kismet has support for using ubertooth, without plugin, I have compiled with success the Kismet Ver 2021 from Git, the WIFI scan is working fine but still the data source of the Ubertooth is not recognized, any idea ? 
Related subject have been posted by myself #476 (under ubertooth topic) and under #402 (under kismet topic)