greatscottgadgets/ubertooth #289: libUSB Error: Pipe error:  (-9)

taijicat, 2017-09-13:
I can't update my firmware
### Steps to reproduce
1.ubertooth-dfu -d bluetooth_rxtx.dfu  -r
2.
3.

### Expected behaviour
upgrade to latest released firmware.

### Actual behaviour
Entering flash programming (DFU) mode
libUSB Error: Pipe error:  (-9)

### Version information
**Operating system**:
Kali Linux VM
**Ubertooth tools version (ubertooth-rx -V):**
libubertooth 1.0 (2017-03-R2), libbtbb 1.0 (2017-03-R2)
**libbtbb version:**

**Ubertooth firmware version (ubertooth-util -v):**
ubertooth git-2c69bb8 (hjoes@hjoes-ThinkPad-T420) Mon Oct 27 19:27:54 CST 2014
If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
```
libUSB Error: Pipe error:  (-9)
Switching to DFU mode...
Unable to find Ubertooth
```
PS: How to add a filter for the usb devices using vmware workstation?

dominicgs, 2017-09-13:
Ubertooth changes USB ID once it enters DFU mode.  If you'rerunning inside a VM you need to make sure that both sets of IDs are passed through.

taijicat, 2017-09-14:
@dominicgs   Thank you for your response,
I had readed previous issues,
I had added the IDs into the filter in  VirtualBox,but  I am still receiving the same error ?

dominicgs, 2017-09-14:
If the Ubertooth is older, you may need to add ffff 0004 to your IDs too.

Alternatively, we have a lot of problems with the USB performance on VirtualBox. Would it be possible to update the Ubertooth from a native OS (boot from a USB stick?) rather than the VM?

taijicat, 2017-09-15:
@dominicgs Thank you for your help.
Finally,I update firmware on raspberry pi.
Although the success of the upgrade, but it is still displayed 
"libUSB Error: Pipe error:  (-9)
Switching to DFU mode
checking firmware signature...
.........
control message unsupported"

dominicgs, 2017-09-15:
I'm not sure where that final `control message unsupported` comes from, but I suspect that it's the reset that we attempt at the end of the update.

Resets over USB are have some interesting timing constraints because we need to return from the USB call before resetting the device.  This appears to be somewhat non-deterministic, but I haven't worked out why.

Anyway, all that is to say that I think it's safe to ignore it, and I'm going to close this issue because we've sorted (via a work around) your initial issue.

closed 2017-09-15 by dominicgs