greatscottgadgets/ubertooth #291: Ubertooth-scan not detecting device that hcitool does

nishant412, 2017-09-27:
### Steps to reproduce
1.Run ubertooth-scan -s -x

### Expected behaviour
It should run hcitool scan using hci0 on my PC followed by an ubertooth-rx and then list the LAP BD_ADDR of devices observed (and if possible resolve the UAP also). I am trying to detect a CSR BC417 based serial-to-bluetooth module (https://www.olimex.com/Products/Components/RF/BLUETOOTH-SERIAL-HC-06/)

### Actual behaviour
While it does list the BD_ADDR of various bluetooth devices, the one device that is paired to it (but not connected) rarely gets listed. I tried doing this test a number of times. The device shows up in the HCI scan almost everytime and yet very rarely does it even show up in ubertooth-rx part of the scan. On the few occasions that it shows up in the scan results, I tried also running ubertooth-rx -l <LAP>, but have never seen UAP resolved. I wanted to ask if you have seen similar behaviour with any other kind of bluetooth devices in the past (Or in case you have tested this particular device out). Also what do you think can be a reason for this behavior? (Maybe I havent correctly understood the algorithm for BlueSniff). I am attaching the output dump from one such scan below.

### Version information
**Operating system: Kali Linux Rolling (and Ubuntu 16.04 LTS)**

**Ubertooth tools version (ubertooth-rx -V) : libubertooth 1.0 (2017-03-R2), libbtbb 1.0 (2017-03-R2)** 

**libbtbb version: **

**Ubertooth firmware version (ubertooth-util -v): 2017-03-R2 (API:1.02)**

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
```
HCI scan
98:D3:32:70:B3:05	stjimmy
Requesting information ...
	BD Address:  98:D3:32:70:B3:05
	Device Name: stjimmy
	LMP Version: 2.1 (0x4) LMP Subversion: 0x100
	Manufacturer: not assigned (6502)
	Features page 0: 0xff 0xff 0x8f 0xfe 0x83 0xe1 0x08 0x80
		<3-slot packets> <5-slot packets> <encryption> <slot offset> 
		<timing accuracy> <role switch> <hold mode> <sniff mode> 
		<park state> <RSSI> <channel quality> <SCO link> <HV2 packets> 
		<HV3 packets> <u-law log> <A-law log> <CVSD> <paging scheme> 
		<power control> <transparent SCO> <broadcast encrypt> 
		<EDR ACL 2 Mbps> <EDR ACL 3 Mbps> <enhanced iscan> 
		<interlaced iscan> <interlaced pscan> <inquiry with RSSI> 
		<extended SCO> <EV4 packets> <EV5 packets> <3-slot EDR ACL> 
		<5-slot EDR ACL> <EDR eSCO 2 Mbps> <EDR eSCO 3 Mbps> 
		<3-slot EDR eSCO> <simple pairing> <extended features> 
	Features page 1: 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00
	Clock offset: 0x0000
AFH disabled.

Ubertooth scan
systime=1506549290 ch= 6 LAP=7978b1 err=0 clk100ns=6233269 clk1=997 s=-44 n=-55 snr=11
systime=1506549291 ch=31 LAP=7978b1 err=1 clk100ns=15279041 clk1=2444 s=-55 n=-55 snr=0
systime=1506549291 ch=48 LAP=1a84bf err=2 clk100ns=16522899 clk1=2643 s=-75 n=-55 snr=-20
systime=1506549292 ch= 9 LAP=4081dd err=0 clk100ns=24950512 clk1=3992 s=-47 n=-55 snr=8
systime=1506549293 ch=70 LAP=1a84bf err=0 clk100ns=38457752 clk1=6153 s=-63 n=-55 snr=-8
systime=1506549294 ch=29 LAP=1a84bf err=0 clk100ns=44858007 clk1=7177 s=-65 n=-55 snr=-10
systime=1506549294 ch=31 LAP=4081dd err=0 clk100ns=46864151 clk1=7498 s=-43 n=-55 snr=12
systime=1506549295 ch= 5 LAP=e828b9 err=2 clk100ns=52430921 clk1=8388 s=-73 n=-55 snr=-18
systime=1506549295 ch=37 LAP=e828b9 err=1 clk100ns=52820079 clk1=8451 s=-71 n=-55 snr=-16
systime=1506549296 ch=15 LAP=e828b9 err=0 clk100ns=62583333 clk1=10013 s=-55 n=-55 snr=0
systime=1506549296 ch=17 LAP=76adc3 err=0 clk100ns=64601839 clk1=10336 s=-63 n=-55 snr=-8
systime=1506549297 ch=40 LAP=4081dd err=0 clk100ns=71865607 clk1=11498 s=-53 n=-55 snr=2
systime=1506549297 ch=59 LAP=7978b1 err=1 clk100ns=75004491 clk1=12000 s=-71 n=-55 snr=-16
systime=1506549298 ch=33 LAP=e828b9 err=0 clk100ns=80670452 clk1=12907 s=-65 n=-55 snr=-10
systime=1506549298 ch=67 LAP=964bac err=0 clk100ns=82869972 clk1=13259 s=-45 n=-55 snr=10
systime=1506549298 ch= 5 LAP=e828b9 err=1 clk100ns=84029767 clk1=13445 s=-45 n=-55 snr=10
systime=1506549298 ch= 5 LAP=e828b9 err=2 clk100ns=84329913 clk1=13493 s=-62 n=-55 snr=-7
systime=1506549298 ch=37 LAP=e828b9 err=0 clk100ns=84419209 clk1=13507 s=-48 n=-55 snr=7
systime=1506549298 ch=37 LAP=e828b9 err=0 clk100ns=84719405 clk1=13555 s=-70 n=-55 snr=-15
systime=1506549298 ch=26 LAP=4081dd err=0 clk100ns=89502612 clk1=14320 s=-46 n=-55 snr=9
systime=1506549299 ch= 4 LAP=4081dd err=0 clk100ns=99175005 clk1=15868 s=-49 n=-55 snr=6
systime=1506549299 ch=68 LAP=1a84bf err=2 clk100ns=99648995 clk1=15943 s=-73 n=-55 snr=-18
systime=1506549300 ch=40 LAP=1a84bf err=0 clk100ns=103372651 clk1=16539 s=-63 n=-55 snr=-8
systime=1506549300 ch=44 LAP=964bac err=0 clk100ns=107318750 clk1=17171 s=-54 n=-55 snr=1
systime=1506549301 ch=31 LAP=1a84bf err=1 clk100ns=110122323 clk1=17619 s=-73 n=-55 snr=-18
systime=1506549301 ch=37 LAP=4081dd err=0 clk100ns=116078047 clk1=18572 s=-51 n=-55 snr=4
systime=1506549302 ch=13 LAP=7978b1 err=0 clk100ns=123798903 clk1=19808 s=-36 n=-55 snr=19
systime=1506549302 ch=17 LAP=76adc3 err=0 clk100ns=127715834 clk1=20435 s=-63 n=-55 snr=-8
systime=1506549302 ch=34 LAP=4081dd err=0 clk100ns=129127503 clk1=20660 s=-51 n=-55 snr=4
systime=1506549303 ch=68 LAP=964bac err=0 clk100ns=131323521 clk1=21011 s=-64 n=-55 snr=-9
systime=1506549303 ch=40 LAP=1a84bf err=1 clk100ns=134859035 clk1=21577 s=-54 n=-55 snr=1
systime=1506549303 ch=40 LAP=7978b1 err=0 clk100ns=135163168 clk1=21626 s=-32 n=-55 snr=23
systime=1506549303 ch=25 LAP=1a84bf err=2 clk100ns=135661742 clk1=21705 s=-48 n=-55 snr=7
systime=1506549303 ch=42 LAP=7978b1 err=0 clk100ns=136837667 clk1=21894 s=-33 n=-55 snr=22
systime=1506549303 ch=76 LAP=7978b1 err=0 clk100ns=139537675 clk1=22326 s=-39 n=-55 snr=16
systime=1506549304 ch=18 LAP=964bac err=0 clk100ns=144521725 clk1=23123 s=-60 n=-55 snr=-5
systime=1506549304 ch=22 LAP=76adc3 err=1 clk100ns=148418871 clk1=23747 s=-68 n=-55 snr=-13
systime=1506549305 ch=41 LAP=4081dd err=0 clk100ns=151717491 clk1=24274 s=-51 n=-55 snr=4
systime=1506549306 ch=27 LAP=76adc3 err=2 clk100ns=169342561 clk1=27095 s=-71 n=-55 snr=-16
systime=1506549307 ch=61 LAP=1a84bf err=0 clk100ns=171662817 clk1=27466 s=-65 n=-55 snr=-10
systime=1506549307 ch= 1 LAP=76adc3 err=1 clk100ns=174968978 clk1=27995 s=-62 n=-55 snr=-7
systime=1506549307 ch= 5 LAP=1a84bf err=0 clk100ns=178862021 clk1=28618 s=-46 n=-55 snr=9
systime=1506549308 ch= 7 LAP=7978b1 err=0 clk100ns=180836719 clk1=28934 s=-43 n=-55 snr=12
systime=1506549309 ch=32 LAP=4081dd err=0 clk100ns=190078921 clk1=30413 s=-41 n=-55 snr=14
systime=1506549309 ch= 2 LAP=76adc3 err=0 clk100ns=191720095 clk1=30675 s=-63 n=-55 snr=-8
systime=1506549309 ch=66 LAP=1a84bf err=0 clk100ns=192462933 clk1=30794 s=-43 n=-55 snr=12
systime=1506549309 ch= 6 LAP=76adc3 err=0 clk100ns=195881143 clk1=31341 s=-61 n=-55 snr=-6
systime=1506549309 ch=40 LAP=4081dd err=0 clk100ns=198180963 clk1=31709 s=-50 n=-55 snr=5

Scan results:
??:??:??:79:78:B1
AFH map: 0xc0200080000500080010
??:??:??:1A:84:BF
AFH map: 0x200000a2000101205400
??:??:??:40:81:DD
AFH map: 0x10020084250300000000
??:??:??:E8:28:B9
AFH map: 0x20800000220000000000
??:??:??:76:AD:C3
AFH map: 0x46004208000000000000
??:??:??:96:4B:AC
AFH map: 0x00000400001000001800
```

dominicgs, 2017-09-29:
Ubertooth only picks up packets from devices that are currently communicating.  If the module is paired but not connected, then it probably isn't sending any packets for the Ubertooth to find.

nishant412, 2017-09-30:
Hi Dominic,

I agree with that and so I ensured that the device is connected and sending data out. 

I'll describe the test I am running . In the case of the module I mentioned, I am running a python script to continuously transmit serial data in an infinite loop. On the master side, I have a laptop connected to the module, and coz it has the SPP , I have a serial port to which I have attached a screen terminal. That way I am certain that data is being sent over Bluetooth.

I see that while ubertooth-scan is able to detect the LAP and resolve the UAP for the master(laptop), I still cant see any packet whatsoever for the slave (HC-06 module). 

Thanks 
Nishant

dominicgs, 2017-10-02:
Ah, now I understand.  You will always see the master device's LAP and UAP but never the slave LAP/UAP.  This is because all packets use the master device's address in the packets, regardless of direction.

closed 2020-12-05 by mikeryan