greatscottgadgets/ubertooth #292: Can not update firmware

nyaz-asti, 2017-09-29:
### Steps to reproduce
1. Ubertooth firmware 534 API:1.00
2. Updating firmware gives me this message non-stop 'control message unsupported'
3. I'm hoping I just need a in between firmware to get the device updated? Like there's 2 or 3 firmware I need to update to, before I can load the latest? Thank you in advance.

dominicgs, 2017-09-29:
If you connect pins 1 & 3 on the [expansion header](https://github.com/greatscottgadgets/ubertooth/wiki/Ubertooth-One#pins-and-leds) as you plug it in, you'll see a little light display from the LEDs.

If you run `ubertooth-dfu` while that's going on, it should work regardless of firmware version.

nyaz-asti, 2017-09-29:
While I did get a different bootup connecting the pins, I'm getting the same message when I try to update the firmware.

nyaz-asti, 2017-09-29:
Nevermind, I got it to work, thanks for your help. The lights dancing didn't last long, so I was quick and got it working when they were briefly. Thanks again.

closed 2017-09-29 by nyaz-asti

dominicgs, 2017-09-29:
I'm glad you got it working.  The timeout on the bootloader is rather short for historic reasons.  I'm sorry that it's a bit misleading.