greatscottgadgets/ubertooth #293: option useless

XingweiLin, 2017-10-03:
### Steps to reproduce
1. `ubertooth-btle -A 38` or `ubertooth-btle -txx:xx:xx:xx:xx:xx`

### Expected behaviour
In `-A`, I want to change the default channel to 38,
In `-t`, I want to only listen to the specific BR_ADDR device

### Actual behaviour
all these options are useless.

### Version information
**Operating system**:
Linux kali 4.12.0-kali1-amd64 #1 SMP Debian 4.12.6-1kali6 (2017-08-30) x86_64 GNU/Linux
In vmware
**Ubertooth tools version (ubertooth-rx -V):**
libubertooth 1.0 (2017-03-R2), 
**libbtbb version:**
libbtbb 1.0 (2017-03-R2)
**Ubertooth firmware version (ubertooth-util -v):**
2017-03-R2

mikeryan, 2017-10-03:
Both options definitely work on the latest firmware. -t will not limit the advertising packets you see, but it will prevent Ubertooth from following connections from any device other than the one you specify.

XingweiLin, 2017-10-04:
But as for the `-A`, when I change use `-A 38`, ubertooth-one still listened on the channel 37.
```
ubertooth-btle -A 38 -f -c test.pcap &> out.txt
```
And as you can see, the results were still in channel 37.

[image]

mikeryan, 2017-10-04:
Are you running the latest host and firmware? Paste the output of ubertooth-util -v

Try resetting the Ubertooth with ubertooth-util -r and see if you still have the same problem.

XingweiLin, 2017-10-04:
```
root@kali:~# ubertooth-util -v
Firmware version: 2017-03-R2 (API:1.02)
```

Ubertooth-one is not stable in my environment.
When I try `ubertooth-util -r`, it failed:
[image]

dominicgs, 2017-10-04:
`libUSB Error: Operation timed out: (-7)` is a fairly common response to resetting the device, because the device resets and disconnects before the response has reached the host system.  This is especially common in virtual machines.

closed 2017-10-26 by XingweiLin