greatscottgadgets/ubertooth #294: could not open Ubertooth device

Deepak111484, 2017-10-30:
### Steps to reproduce
1.
2.
3.

I am beginner to ubertooth project. After installation of possible dependencies then I connected ubertooth one to Kali-linux(host)  not sure if ubertooth one has firmware installed

### Expected behaviour
should be detect and run the ubertooth one commands 
### Actual behaviour 
could not open Ubertooth device; 
unable to detect in lsusb even tried with inserted pin as shown in picture https://github.com/greatscottgadgets/ubertooth/wiki/Ubertooth-One#pins-and-leds (can see 5 leds- 2 green, 3 orange & umber)
 for libbtbb-2015-10-R1 
~/libbtbb-2015-10-R1/build/ubertooth-2015-10-R1/host/build/spectools# ubertooth-util -v
ubertooth-util: error while loading shared libraries: libubertooth.so.1: cannot open shared object file: No such file or directory

### Version information
**Operating system**:
Kali-Linux
**Ubertooth tools version (ubertooth-rx -V):**
libubertooth 0.2 (GITDIR-NOTFOUND), libbtbb 0.3 (GITDIR-NOTFOUND)
**libbtbb version:**
ubertooth-2017-03-R2
**Ubertooth firmware version (ubertooth-util -v):**
could not open Ubertooth device

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
```
root@kali:~/ubertooth-2017-03-R2# ubertooth-rx -V
libubertooth 0.2 (GITDIR-NOTFOUND), libbtbb 0.3 (GITDIR-NOTFOUND)
root@kali:~/ubertooth-2017-03-R2# ubertooth-util -v
could not open Ubertooth device
ubertooth-util - command line utility for Ubertooth Zero and Ubertooth One
Usage:
	-a[0-7] get/set power amplifier level
	-b get hardware board id number
	-c[2400-2483] get/set channel in MHz
	-C[0-78] get/set channel
	-d[0-1] get/set all LEDs
	-e start repeater mode
	-f activate flash programming (DFU) mode
	-h display this message
	-i activate In-System Programming (ISP) mode
	-I identify ubertooth device by flashing all LEDs
	-l[0-1] get/set USR LED
	-m display range test result
	-n initiate range test
	-p get microcontroller Part ID
	-q[1-225 (RSSI threshold)] start LED spectrum analyzer
	-r full reset
	-s get microcontroller serial number
	-S stop current operation
	-t intitiate continuous transmit test
	-U<0-7> set ubertooth device to use
	-v get firmware revision number
	-V get compile info
	-z set squelch level
-------------------------------------------------------------------------------------------------------------
~/libbtbb-2015-10-R1/build/ubertooth-2015-10-R1/host/build/spectools# ubertooth-util -v
ubertooth-util: error while loading shared libraries: libubertooth.so.1: cannot open shared object file: No such file or directory

--

....

Deepak111484, 2017-10-30:
corrections- my machin do not have bluetooth adaptor :p

but pls check all other data hope all are proper or pls coreect

mikeryan, 2017-10-30:
Run sudo ldconfig and retry

Deepak111484, 2017-10-31:
Now I have blue adapter detected in kali (rolling) in VM, not able to detect ubertooth one;
(libbtbb-2015-10-R1 & ubertooth-2015-10-R1)

root@kali:~# cat /etc/udev/rules.d/40-ubertooth.rules 
# Possible Ubertooth DFU
ATTRS{idVendor}=="ffff", ATTRS{idProduct}=="0004", MODE="0660", GROUP="plugdev"
# Ubertooth Zero
ATTRS{idVendor}=="1d50", ATTRS{idProduct}=="6000", MODE="0660", GROUP="plugdev"
# Ubertooth Zero DFU
ATTRS{idVendor}=="1d50", ATTRS{idProduct}=="6001", MODE="0660", GROUP="plugdev"
# Ubertooth One
ATTRS{idVendor}=="1d50", ATTRS{idProduct}=="6002", MODE="0660", GROUP="plugdev"
# Ubertooth One DFU
ATTRS{idVendor}=="1d50", ATTRS{idProduct}=="6003", MODE="0660", GROUP="plugdev"

kennethblite, 2018-03-08:
Try checking version with ubertooth-dfu. You may have a firmware versioning error, it may tell you to upgrade your firmware to the newest version.

MatteoBaldini, 2018-08-02:
Hello everyone

I have a very similar problem and I'm a noob as well. Whenever I try to run ubertooth-utils -v to get the version the output is "could not open Ubertooth device" followed by all the possible options, none of which works. The device is a UD100 adapter by the way.
I followed the tutorial and never had any errors at any point. If I run blue_hydra  it confirms that there's no ubertooth device.
Any idea as to why? All help is appreciated.

Deepak111484, 2018-08-02:
Hi,

That time only I have sorted out the issue by updating kali Linux with the
latest package. Please verify your build.

All the best.

Thanks,
Deepak

mikeryan, 2018-08-02:
Do you have an Ubertooth device? The UD100 is not an Ubertooth, so the tools will not work with it.

philosofonusus, 2019-05-20:
You need to install libbtbb

git clone https://github.com/greatscottgadgets/libbtbb
cd libbtbb
mkdir build
cd build
cmake ..
make
sudo make install
cmake -DINSTALL_DIR=/path/to/install -DINCLUDE_DIR=/path/to/include ..

closed 2020-12-05 by mikeryan