greatscottgadgets/ubertooth #295: Ubertooth-follow showing failed to decode

sandeep1parmar, 2017-11-08:
### Steps to reproduce
1) use the command "sudo ubertooth-follow -l LAP -u UAP "
for example "sudo ubertooth-follow -l b3a1ec -u ee"

### Expected behaviour
ubertooth-follow utility should follow the master connection with defined LAP and UAP, and should decode the packets.

### Actual behaviour
ubertooth follow is continuously showing  "Failed to decode packet"

### Version information
Ubuntu 16.04

**Ubertooth tools version (ubertooth-rx -V):**
libubertooth 1.0 (2017-03-R2), 

**libbtbb version:**
libbtbb 1.0 (2017-03-R2)

**Ubertooth firmware version (ubertooth-util -v):**
Firmware version: 2017-03-R2 (API:1.02)

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
$ sudo ubertooth-follow -l b3a1ec -u ee
TESTTTT Clock:    0x324b27f
Address given, assuming address is remote
Address: 00:00:EE:B3:A1:EC
Not use AFH
systime=1510137046 ch=52 LAP=b3a1ec err=0 clkn=52784478 clk_offset=1661 s=0 n=-55 snr=55
offset < CLK_TUNE_TIME
CLK100ns Trim: 5661
systime=1510137046 ch=70 LAP=b3a1ec err=0 clkn=52785660 clk_offset=2275 s=-38 n=-55 snr=17
Failed to decode packet
systime=1510137046 ch=24 LAP=b3a1ec err=2 clkn=52785980 clk_offset=2290 s=-53 n=-55 snr=2
Failed to decode packet
systime=1510137049 ch=24 LAP=b3a1ec err=2 clkn=52796386 clk_offset=2595 s=-77 n=-55 snr=-22
systime=1510137059 ch=33 LAP=b3a1ec err=1 clkn=52827719 clk_offset=3497 s=0 n=-55 snr=55
systime=1510137060 ch= 4 LAP=b3a1ec err=0 clkn=52829791 clk_offset=3564 s=0 n=-55 snr=55
systime=1510137122 ch=35 LAP=b3a1ec err=0 clkn=53028810 clk_offset=2823 s=-29 n=-55 snr=26
offset > CLK_TUNE_TIME
CLK100ns Trim: 573
Clock drifted 573 in 76.353750 s. 0 PPM too fast.
systime=1510137122 ch=68 LAP=b3a1ec err=2 clkn=53028876 clk_offset=2265 s=0 n=-55 snr=55
Failed to decode packet
systime=1510137122 ch=65 LAP=b3a1ec err=0 clkn=53028914 clk_offset=2259 s=-29 n=-55 snr=26
Failed to decode packet
systime=1510137122 ch=25 LAP=b3a1ec err=0 clkn=53028996 clk_offset=2262 s=-68 n=-55 snr=-13
Failed to decode packet
systime=1510137122 ch=18 LAP=b3a1ec err=0 clkn=53029042 clk_offset=2257 s=-29 n=-55 snr=26
Failed to decode packet
systime=1510137122 ch=53 LAP=b3a1ec err=0 clkn=53029132 clk_offset=2279 s=0 n=-55 snr=55
Failed to decode packet
systime=1510137122 ch=46 LAP=b3a1ec err=1 clkn=53029264 clk_offset=2274 s=-69 n=-55 snr=-14
Failed to decode packet
systime=1510137122 ch=16 LAP=b3a1ec err=1 clkn=53029292 clk_offset=2289 s=-31 n=-55 snr=24
Failed to decode packet
systime=1510137122 ch=28 LAP=b3a1ec err=2 clkn=53029352 clk_offset=2274 s=-65 n=-55 snr=-10
Failed to decode packet
systime=1510137136 ch=58 LAP=b3a1ec err=1 clkn=53074599 clk_offset=3574 s=-36 n=-55 snr=19
systime=1510137136 ch=71 LAP=b3a1ec err=2 clkn=53074685 clk_offset=3589 s=-51 n=-55 snr=4
systime=1510137136 ch=11 LAP=b3a1ec err=2 clkn=53074727 clk_offset=3575 s=-38 n=-55 snr=17
systime=1510137136 ch=23 LAP=b3a1ec err=0 clkn=53075077 clk_offset=3601 s=0 n=-55 snr=55
offset > CLK_TUNE_TIME
CLK100ns Trim: 1351
Clock drifted 1351 in 14.458438 s. 9 PPM too fast.
systime=1510137136 ch=20 LAP=b3a1ec err=0 clkn=53075174 clk_offset=2230 s=-39 n=-55 snr=16
Failed to decode packet

Regards
Sandeep Parmar

sandeep1parmar, 2017-11-17:
Hi dominic,
As mentioned earlier Main aim is to understand whether ubertooth can decode FHS Packets successfully for Bluetooth classic.
Doing some further progress, we are able to get some packets decoded successfully for certain test runs as below:
systime=1510817213 ch=68 LAP=b3a1ec err=0 clkn=4417284 clk_offset=2457 s=0 n=-55 snr=55
systime=1510817214 ch=47 LAP=b3a1ec err=0 clkn=4418752 clk_offset=2440 s=0 n=-55 snr=55
Test Packet Type is: 1Test btbb_decode_header data 1 Packet decoded with clock 0x20 (rv=1)
  Type: POLL
Test data 1   Type: POLL
systime=1510817214 ch=58 LAP=b3a1ec err=0 clkn=4419108 clk_offset=2478 s=0 n=-55 snr=55
systime=1510817215 ch=69 LAP=b3a1ec err=0 clkn=4421388 clk_offset=2467 s=0 n=-55 snr=55
systime=1510817215 ch=36 LAP=b3a1ec err=0 clkn=4422442 clk_offset=2442 s=-28 n=-55 snr=27
Test data 0 Failed to decode packet
systime=1510817215 ch=44 LAP=b3a1ec err=0 clkn=4422670 clk_offset=2443 s=-21 n=-55 snr=34
Test data 0 Failed to decode packet
systime=1510817216 ch=73 LAP=b3a1ec err=0 clkn=4424640 clk_offset=2467 s=0 n=-55 snr=55
systime=1510817216 ch=77 LAP=b3a1ec err=1 clkn=4426516 clk_offset=2493 s=0 n=-55 snr=55
systime=1510817216 ch=60 LAP=b3a1ec err=0 clkn=4427134 clk_offset=2461 s=-16 n=-55 snr=39
systime=1510817217 ch=47 LAP=b3a1ec err=0 clkn=4428320 clk_offset=2491 s=0 n=-55 snr=55
offset > CLK_TUNE_TIME
CLK100ns Trim: 241
Clock drifted 241 in 46.716250 s. 0 PPM too fast.
systime=1510817217 ch=77 LAP=b3a1ec err=0 clkn=4428988 clk_offset=2244 s=0 n=-55 snr=55
Test data 0 Failed to decode packet
systime=1510817217 ch=34 LAP=b3a1ec err=2 clkn=4429760 clk_offset=2245 s=0 n=-55 snr=55
Test Packet Type is: 1Test btbb_decode_header data 1 Packet decoded with clock 0x20 (rv=1)
  Type: POLL
Test data 1   Type: POLL
systime=1510817217 ch=71 LAP=b3a1ec err=0 clkn=4430208 clk_offset=2242 s=0 n=-55 snr=55
Test Packet Type is: 1Test btbb_decode_header data 1 Packet decoded with clock 0x00 (rv=1)
  Type: POLL
Test data 1   Type: POLL
systime=1510817218 ch=76 LAP=b3a1ec err=0 clkn=4431244 clk_offset=2258 s=0 n=-55 snr=55
Test data 0 Failed to decode packet
systime=1510817218 ch=78 LAP=b3a1ec err=0 clkn=4431308 clk_offset=2252 s=0 n=-55 snr=55

**However problem faced is wireshark logs show FHS Packets but ubertooth console logs when ubertooth-follow application is run, does not show FHS Packets.
We want to understand why such discrepancy between wireshark and ubertooth-follow console logs.
Are we missing something??**

Awaiting your response soon.

**Attaching log files for your reference.**
[2.pcapng.zip](https://github.com/greatscottgadgets/ubertooth/files/1482315/2.pcapng.zip)
[2_ubertooth_Logs.txt](https://github.com/greatscottgadgets/ubertooth/files/1482314/2_ubertooth_Logs.txt)

regards
sandeep

mikeryan, 2020-12-05:
Check out the new btbr firmware, which will be merged in the next release.

closed 2020-12-05 by mikeryan