greatscottgadgets/ubertooth #296: libUSB Error: Timeout: (-7)

ykram, 2017-12-04:
### Steps to reproduce
1. ubertooth-util -v

### Expected behaviour
Expect to see firmware on device

### Actual behaviour
Receive an error message:

libUSB Error: Timeout:  (-7)
Firmware revision: error: -7

### Version information
Kali 4.13.0-amd64

**Ubertooth tools version (ubertooth-rx -V):**
libubertooth 0.2

**libbtbb version:**
libbtbb 0.3

**Ubertooth firmware version (ubertooth-util -v):**
N/A

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output

Using `lsusb` after initially inserting the device shows:

`iManufacturer        1 Great Scott Gadgets`
`iProduct                 2 Ubertooth One`
`iSerial                     3 <omitted>`
...
`Device Status:         0x0000`
`  (Bus Powered)`

However, after I try to initiate a `ubertooth-util -v`, I receive an error saying that libUSB timed out (-7).

`lsusb -vd 1d50:6002` then shows that

`iManufacturer`, `iProduct` and `iSerial` are all showing as (Error) but the `Device Status` is still 0x0000.

mikeryan, 2017-12-04:
Are you running Kali inside a VM? How many LEDs are lit, and what color are they?

ykram, 2017-12-04:
This Kali being run via USB, not in a VM. I have 3 LEDs lit, 2 green and 1 red.

dominicgs, 2017-12-04:
Do any of the other functions work?  e.g. `ubertooth_util -l`?  It should report the status of the LEDs.

ykram, 2017-12-04:
Unfortunately, they do not. I get:

libUSB Error: Timeout:  (-7)
USR LED Status: -7

dominicgs, 2017-12-04:
How about any of the other tools? e.g.'ubertooth-rx`?

Are you able to run `ubertooth-dfu` to update the firmware to the latest version?

ykram, 2017-12-04:
`ubertooth-rx` and the other tools also return libUSB error -7.

`ubertooth-dfu -d bluetooth_rxtx.dfu -r` gives me the following:

libUSB Error: Operation timed out:  (-7)
Switching to DFU mode...
Unable to find Ubertooth

ykram, 2017-12-04:
I tried installing the tools on a separate linux laptop running Ubuntu 16.04.3 and while I still get the -7 error from libUSB, `ubertool-util -v` gives me somewhat more information:

`libUSB Error: Operation timed out:  (-7)`
`Firmware version: error: -7 (API:1.02)`

dominicgs, 2017-12-04:
I think we'll need to force your Ubertooth in to bootloader mode.  If you have a paper clip you can connect two pins on the expansion header: https://pbs.twimg.com/profile_banners/19587312/1354079866/1500x500

You'll need to run the `ubertooth-dfu -d bluetooth_rxtx.dfu -r` command while the lights are still flashing.

ykram, 2017-12-04:
That looks like it did the trick! ubertooth-util shows the proper version
and capturing in WireShark world as expected. Thank you!

closed 2017-12-04 by ykram