greatscottgadgets/ubertooth #297: ubertooth-btle errors out after ~30 seconds

TomSoderling, 2017-12-09:
### Steps to reproduce
1. Following these steps, I created a pipe and added it as an interface in wireshark. https://github.com/greatscottgadgets/ubertooth/wiki/Capturing-BLE-in-Wireshark
2. Run command in terminal: ubertooth-btle -p -c /tmp/pipe
3. Watch packets stream into Wireshark

### Expected behaviour
My expectation is that this command should keep running indefinitely, sniffing BLE packets

### Actual behaviour
Packets appear in the terminal window and in wireshark for around 22 to 30 seconds and then will stop suddenly and show this error in terminal:

libUSB Error: Input/Output Error:  (-1)
USB error

### Version information
**Operating system**:
MacOS High Sierra, version 10.13.1 (17B1003)

**Ubertooth tools version (ubertooth-rx -V):**
libubertooth 1.0 (2017-03-R2), libbtbb 1.0 (2017-03-R2) 

**libbtbb version:**
libbtbb-2017-03-R2  

**Ubertooth firmware version (ubertooth-util -v):**
Firmware version: 2017-03-R2 (API:1.02) 

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.
Wireshark Version 2.4.2 (v2.4.2-0-gb6c63ae. 

### Output
```
Insert any commandline or build output here
```

TomSoderling, 2017-12-09:
I get the same error after 15-60 seconds using the -f option instead: ubertooth-btle -f -c /tmp/pipe

mikeryan, 2018-06-27:
Please try the latest pre-release 2018-06-R1 and reopen the issue if you are still having this problem.

closed 2018-06-27 by mikeryan

zhangyoufu, 2019-03-19:
I'm experiencing the same symptom.

HW: MacBookPro12,1 (USB Type-A)
OS: Mac OS X 10.11.6 (15G22010)
libusb: 1.0.22

```
# ubertooth-util -v
Firmware version: 2018-12-R1 (API:1.06)

# ubertooth-util -V
ubertooth 2018-12-R1 (mikeryan@steel) Tue Dec  4 22:23:44 PST 2018
```

After ubertooth-btle quit with `libUSB Error: Input/Output Error:  (-1)`, libusb cannot enumerate this device until I detach and re-attach Ubertooth. This may be related to https://github.com/libusb/libusb/issues/290