greatscottgadgets/ubertooth #298: ubertooth-scan error? error Unable to find hci0(1)

sikongsima, 2017-12-21:
Hey,
when I try Use the command “**ubertooth-scan**”，has a error “**error Unable to find hci0(1)**”,use "**hcitool scan**",has a error "**No such device**",
only “ ubertooth-btle -f -c ”，Can be successful  ，I using a Ubuntu VM，The installed version is Release 2015 10 R1，ubertooth-one is 2C03 ，What may be the reason for this？Thanks very much！！！

dominicgs, 2017-12-23:
Ubertooth scan uses an off the shelf Bluetooth device in conjunction with
the Ubertooth, you don't appear to have one connected.

If you run `ubertooth-btle -z` it will do something similar without the
Bluetooth dongle.

<https://github.com/greatscottgadgets/ubertooth/issues/298>, or mute the
thread

dominicgs, 2017-12-23:
Oops, I meant `ubertooth-rx -z`

sikongsima, 2017-12-26:
Use the command “**ubertooth-btle -z**”，has a error “USB error libusb error：timeout(7)”,What's the reason for this? @troth  @dominicgs

dominicgs, 2017-12-26:
`ubertooth-rx -z`

EMCP, 2018-01-02:
On latest drivers + firmware, I am also getting an error with `ubertooth-scan` , but the `ubertooth-specscan` seems to be working

```
pi@raspberrypi:~/Dev/ubertooth-2017-03-R2/host/kismet/plugin-ubertooth $ ubertooth-scan
error: Unable to find hci0 (-1)

...

pi@raspberrypi:~/Dev/ubertooth-2017-03-R2/host/kismet/plugin-ubertooth $ ubertooth-rx -z
systime=1514908483 ch=74 LAP=f5859c err=0 clkn=324196 clk_offset=704 s=-63 n=-55 snr=-8
systime=1514908484 ch=78 LAP=f2f8f5 err=0 clkn=325469 clk_offset=6183 s=-66 n=-55 snr=-11
systime=1514908484 ch=20 LAP=f5859c err=1 clkn=327148 clk_offset=697 s=-51 n=-55 snr=4
systime=1514908484 ch= 5 LAP=de45be err=2 clkn=327375 clk_offset=6107 s=-62 n=-55 snr=-7
```

closed 2020-12-05 by mikeryan