greatscottgadgets/ubertooth #299: Kismet does not find ubertooth

Bre5933, 2017-12-28:
### Steps to reproduce
1. Followed build guide
2. Followed README in /host/kismet/plugin-ubertooth
3. Run kismet as root and add source

### Expected behaviour
Ubertooth device should be detected and added to sources

### Actual behaviour
Failed to find type for auto-type source "ubertooth".....

### Version information
**Operating system**: Ubuntu 14.04 LTS

**Ubertooth tools version (ubertooth-rx -V):**
libubertooth 1.0 (2017-03-R2), libbtbb 1.0 (git-1b6084d)

**libbtbb version:**
libubertooth 1.0 (2017-03-R2), libbtbb 1.0 (git-1b6084d)

**Ubertooth firmware version (ubertooth-util -v):**
Firmware version: 2017-03-R2 (API:1.02)

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.
Kismet 2013-03-R0
### Output
[image]

Bre5933, 2017-12-28:
I should add that the device is connected. I can still use the ubertooth spectrum analyzer

Bre5933, 2017-12-28:
Fixed the issue with a little help from this Hak5 video. 
https://www.google.com/search?client=ubuntu&channel=fs&q=using+kismet+with+ubertooth&ie=utf-8&oe=utf-8#kpvalbx=1
the only issue is one the kismet dir is moved you still need to:
./configure
make && make plugins
sudo make suidinstall
sudo make plugins-install
Add "pcapbtbb" to the "logtypes=..." line in kismet.conf

then cd  to /host/kismet/plugin-ubertooth

and finally make && make install

after that i ran kismet from home directory and added the ubertooth source

EMCP, 2018-01-02:
I am having similar issues.  Tried latest Kismet source code installation, which failed with the known header issue.. However rolling back to the instructions 2013 Kismet yielded nothing working, no sources detected in Kismet.

You say you got it working by `then cd to /host/kismet/plugin-ubertooth`

but there is no make file in that directory ..

```
pi@raspberrypi:~/Dev/ubertooth-2017-03-R2/host/kismet/plugin-ubertooth $ make && make install
Makefile:7: /usr/src/kismet/Makefile.inc: No such file or directory
make: *** No rule to make target '/usr/src/kismet/Makefile.inc'.  Stop.
```

Bre5933, 2018-01-02:
I believe you may need to edit the make file in host/kismet/plugin-ubertooth to point at that directory instead of /usr/src/kismet. I'm currently don't have access to the PC I used. Could help you tomorrow though

ramamr33, 2019-07-17:
Hi All,

I have tried to install the kismet plugin-ubertooth downloaded from Github (https://github.com/greatscottgadgets/ubertooth) but I am still getting the same error:

[image]

I have tried to run make && make install plugins command on the kismet src directory (downloaded from GitHub - https://github.com/kismetwireless/kismet.git) but I am getting stuck with this error as well:

make: *** No targets specified and no makefile found.  Stop.

I confirm 2 files **MakeFile.in** and **MakeFile.inc.in** both exist on the kismet src directory:
[image]

Please kindly assist.
Thanks.

mikeryan, 2020-12-05:
Legacy Kismet is no longer supported by this project.

closed 2020-12-05 by mikeryan