greatscottgadgets/ubertooth #300: Installing firmware error, troubleshooting instruction not working

EMCP, 2017-12-29:
### Steps to reproduce
1. Performed the installation with latest release successfully on 16.04 LTS
2. `lsusb` is not showing the ubertooth
3. when attempting to update firmware, got errors

### Expected behaviour
Tell us what you expect should happen

no errors, or troubleshooting to work

### Actual behaviour
Tell us what happens instead

```
ubertooth-one-firmare-bin$ ubertooth-dfu -d bluetooth_rxtx.dfu -r
libUSB Error: Input/Output Error: (-1)
Switching to DFU mode...
....
...
Detached
```

then I tried

```
ubertooth-one-firmare-bin$ make clean all && make && ubertooth-dfu -r -d bluetooth_rxtx.dfu
make: *** No rule to make target 'clean'.  Stop.

```

### Version information
**Operating system**:
`Ubuntu 16.04`
**Ubertooth tools version (ubertooth-rx -V):**
`libubertooth 1.0 (2017-03-R2), libbtbb 1.0 (2017-03-R2)`

**libbtbb version:**

**Ubertooth firmware version (ubertooth-util -v):**

`could not open Ubertooth device`

EMCP, 2017-12-29:
i reset the machine and retried... i guess ´´`ldconfig` wasn't enough?

closed 2017-12-29 by EMCP

benmargolin, 2019-10-18:
I had the same issue and unplugging/replugging the ubertooth device physically, resolved it without a machine reboot.

attila123, 2022-09-13:
I also needed to reboot my system for the same reason. After reboot, the firmware upgrade went fine.  
BTW I use Arch :)