greatscottgadgets/ubertooth #301: Update Firmware Failed - control message unsupported

pvidalis, 2018-01-06:
Hi, 
I tried to install ubertooth by following the suggested procedure but when I'm running the command to update firmware I'm receiving the following message

`pi@raspberrypi:~/ubertooth-2017-03-R2/ubertooth-one-firmware-bin $ ubertooth-dfu -d bluetooth_rxtx.dfu -r
Switching to DFU mode...
Checking firmware signature
........................................
........................................
................
control message unsupported
`
I have also tried this in kubuntu 17.10, with the same results
Thanks in advance

### Steps to reproduce
1. Install Release 2017-03-R2 according to Build Guide
2. Update Firmware

### Expected behaviour
I'm expecting to see messages similar to this
`Checking firmware signature
No DFU devices found - attempting to find Ubertooth devices

1) Found 'Ubertooth One' with address 0x1d50 0x6002

Select a device to flash (default:1, exit:0):`

### Actual behaviour
`ubertooth-dfu -d bluetooth_rxtx.dfu -r
Switching to DFU mode...
Checking firmware signature
........................................
........................................
................
control message unsupported`

### Version information
**Operating system**:  Raspberry pi 3
`pi@raspberrypi:~ $ cat /etc/debian_version
8.0

pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.4.21-v7+ #911 SMP Thu Sep 15 14:22:38 BST 2016 armv7l GNU/Linux
`
**Ubertooth tools version (ubertooth-rx -V):**
`pi@raspberrypi:~ $ ubertooth-rx -V
libubertooth 1.0 (2017-03-R2), libbtbb 1.0 (2017-03-R2)
`
**libbtbb version:**
libbtbb-2017-03-R2.tar.gz

**Ubertooth firmware version (ubertooth-util -v):**
`pi@raspberrypi:~ $ ubertooth-util -v
Firmware version: 2017-03-R2 (API:1.02)

pi@raspberrypi:~ $ ubertooth-util -V
ubertooth 2017-03-R2 (dominicgs@hydrogen) Mon Mar 13 16:09:02 MDT 2017
`
If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
The process went smooth just those warnings appear.
```
pi@raspberrypi:~/libbtbb-2017-03-R2/wireshark/plugins/btbb/build $ make
[ 20%] Generating plugin.c
Scanning dependencies of target btbb
[ 40%] Building C object CMakeFiles/btbb.dir/plugin.c.o
[ 60%] Building C object CMakeFiles/btbb.dir/packet-btbb.c.o
[ 80%] Building C object CMakeFiles/btbb.dir/packet-btbrlmp.c.o
/home/pi/libbtbb-2017-03-R2/wireshark/plugins/btbb/packet-btbrlmp.c: In function ‘proto_register_btbrlmp’:
/home/pi/libbtbb-2017-03-R2/wireshark/plugins/btbb/packet-btbrlmp.c:2262:4: warning: large integer implicitly truncated to unsigned type [-Woverflow]
    FT_UINT64, BASE_HEX, NULL, 0x0000ffffffffffff,
    ^
/home/pi/libbtbb-2017-03-R2/wireshark/plugins/btbb/packet-btbrlmp.c:2267:4: warning: large integer implicitly truncated to unsigned type [-Woverflow]
    FT_UINT64, BASE_HEX, NULL, 0x0000ffffffffffff,
    ^
/home/pi/libbtbb-2017-03-R2/wireshark/plugins/btbb/packet-btbrlmp.c:2272:4: warning: large integer implicitly truncated to unsigned type [-Woverflow]
    FT_UINT64, BASE_HEX, NULL, 0x0000ffffffffffff,
    ^
[100%] Linking C shared module btbb.so
[100%] Built target btbb
```
[raspberry.txt](https://github.com/greatscottgadgets/ubertooth/files/1608677/raspberry.txt)

dominicgs, 2018-01-07:
The firmware update went smoothly, but the output doesn't match the guide.

I'll update the guide before I close this issue.

pvidalis, 2018-01-08:
OK, 
thanks a lot.

BG7YWL, 2020-07-28:
@dominicgs 
I have the same problem.
[image]

closed 2020-12-05 by mikeryan