greatscottgadgets/ubertooth #302: Not generating desired packets through wireshark

apeturis, 2018-03-09:
### Steps to reproduce
1. Perform /tmp/pipe scan using wireshark
2. Attempt to sync wiimote with wii
3. See packets between wiimote and wii

### Expected behaviour
I was expecting to see the packets displaying the connection between the wiimote and wii. I am under the impression the wiimote is an HID device using HCI authentication and I would like to observe the packets to reverse engineer a wiimote.

### Actual behaviour
No packets appear involving the wiimote or wii's BT Address.

### Version information
**Operating system**:
Linux 9.3 Stretch
**Ubertooth tools version (ubertooth-rx -V):**

**libbtbb version:**
2017-03-R2
**Ubertooth firmware version (ubertooth-util -v):**
2017-03-R2
If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.
Wireshark 2.4.6
### Output
```
Insert any commandline or build output here
```
ubertooth-btle -f -c /tmp/pipe

LarryCheech, 2018-03-15:
mkfifo /tmp/<dir>
wireshark -k -i /tmp/<dir> &
ubertooth-btle -f -c /tmp/<dir> &

Hope this helps.

apeturis, 2018-03-15:
I will test it as soon as I get home and let you know, thanks.

mikeryan, 2018-03-15:
Wiimotes use classic BT, not BLE. Our classic BT sniffing support is extremely limited. You will not be able to capture any traffic from Wiimotes using Ubertooth.

apeturis, 2018-03-15:
The last command I recall using is the ubertooth rx command to listen to packets and it would record me clicking the 1 and 2 buttons to sync, but I didn't see an authentication request. Is that process something that is not going to be captured from the ubertooth?

mikeryan, 2018-03-15:
You likely saw the device making itself discoverable and sending out numerous packets on the inquiry scan or page scan transports. You most likely won't be able to capture any meaningful connection data from the Wiimote.

apeturis, 2018-03-15:
Thank you for the replies I am going to use the data you have provided and try a few more things. I am ok with the thread closing now as my issue has been answered.

Thanks again.

closed 2018-03-15 by apeturis