greatscottgadgets/ubertooth #306: Bricked ubertooth after firmware upgrade?

krisclarkdev, 2018-04-25:
### Steps to reproduce
1. Install libbtbb and ubertooth-tools via [these steps](https://github.com/greatscottgadgets/ubertooth/wiki/Build-Guide)
2. Build firmware via [these steps](https://github.com/greatscottgadgets/ubertooth/wiki/Firmware)
3. Upgrade firmware via

```
ubertooth-dfu -d bluetooth_rxtx.dfu -r
```

4. After successful upgrade unplug the Ubertooth One and plug back in
5. Check firmware version

### Expected behaviour
I should be able to access the firmware version of my Ubertooth One

### Actual behaviour
What's actully happening is ubertooth-tools is unable to find/open the Ubertooth One using ANY provided commands.  All I get is

```
could not open Ubertooth device
Unable to find Ubertooth
```

### Version information
**Operating system**:
Ubuntu 18.04 LTS

**Ubertooth tools version (ubertooth-rx -V):**
libubertooth 1.0 (2017-03-R2)

**libbtbb version:**
libbtbb 1.0 (2017-03-R2)

**Ubertooth firmware version (ubertooth-util -v):**
could not open Ubertooth device

### Output
```
lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 8087:07dc Intel Corp. 
Bus 001 Device 003: ID 0bda:0129 Realtek Semiconductor Corp. RTS5129 Card Reader Controller
Bus 001 Device 002: ID 0c45:6712 Microdia 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

dmesg
***nothing relevent***
```

dominicgs, 2018-05-01:
It certainly looks like there was a problem with the firmware update.  Not to worry, we can recover it with a paperclip.

 * Connect pins 1 and 3 on the expansion header, as shown here: https://pbs.twimg.com/profile_banners/19587312/1354079866/1500x500
 * There should be an LED light show to signal that you're in bootloader mode, there's a relatively narrow timeout on this, so you'll need to run the firmware update quickly, or unplug and replug the Ubertooth
 * Connect the Ubertooth to your system and update the firmware using `ubertooth-dfu -d bluetooth_rxtx.dfu -r`

krisclarkdev, 2018-05-11:
Thanks for the reply @dominicgs I've

* Connected pins 1 and 3 as shown in your photo
* I then see the device in bootloader mode
* Right after plugging in the device I run the firmware update and it's successful as far as I can tell

`Checking firmware signature
........................................
........................................
............................
Detached`

I then removed the device and reconnected, when I run

`ubertooth-util -v`
or
`sudo ubertooth-util -v`

I still get the following

`
ubertooth/firmware/bluetooth_rxtx$ ubertooth-util -v
could not open Ubertooth device
ubertooth-util - command line utility for Ubertooth Zero and Ubertooth One

Common options:
	-v get firmware revision number
	-V get compile info
	-I identify ubertooth device by flashing all LEDs
	-d[0-1] get/set all LEDs
	-l[0-1] get/set USR LED
	-S stop current operation
	-r full reset
	-U<0-7> set ubertooth device to use

Radio options:
	-a[0-7] get/set power amplifier level
	-c[2400-2483] get/set channel in MHz
	-C[0-78] get/set channel
	-q[1-225 (RSSI threshold)] start LED spectrum analyzer
	-t intitiate continuous transmit test
	-z set squelch level

Range test:
	-e start repeater mode
	-m display range test result
	-n initiate range test

Miscellaneous:
	-f activate flash programming (DFU) mode
	-i activate In-System Programming (ISP) mode
	-b get hardware board id number
	-p get microcontroller Part ID
	-s get microcontroller serial number
`

dominicgs, 2018-05-12:
That's very strange, it certainly looks like a successful firmware update, but clearly something is going wrong.

After the firmware update, when the device has reset, how many LEDs on it are lit?

Could you share the output from `dmesg` when you connect the Ubertooth?  (I like to use `dmesg -w` for this), and could you also give me the output from `lsusb`?

dominicgs, 2018-05-12:
Also, which firmware file did you write to the Ubertooth?

dgrizzle, 2018-05-29:
I am experiencing the exact same problem that the OP described. 

**Operating System**
Ubuntu 18.04 LTS

**ubertooth-rx -V**
`libubertooth 1.0 (git-1e45cf2*), libbtbb 1.0 (git-1b6084d)`

**lsusb**
```
Bus 002 Device 002: ID 8087:8000 Intel Corp. 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:8008 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 0a5c:216f Broadcom Corp. BCM20702A0 Bluetooth
Bus 003 Device 002: ID 04f3:21e5 Elan Microelectronics Corp. 
Bus 003 Device 015: ID 0403:6001 Future Technology Devices International, Ltd FT232 USB-Serial (UART) IC
Bus 003 Device 004: ID 0bda:573c Realtek Semiconductor Corp. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
Attempted to write firmware file built from latest commit: 1e45cf2

dominicgs, 2018-05-29:
@dgrizzle Did you try connecting the pins as I suggested in a previous comment (https://github.com/greatscottgadgets/ubertooth/issues/306#issuecomment-385673357)?  Did it go in to the bootloader mode with the LED flashing pattern?

dgrizzle, 2018-05-29:
Yes, I did that, and got the same outcome as the OP. 

`Checking firmware signature ........................................ ........................................ ............................ Detached`

I then unplug/plug in and RSTLED turns on but no new information in dmesg -w

I've tried shorting the pins and running ubertooth-dfu multiple times with the same outcome.
The device does not enumerate on USB at all- neither as a bootloader nor as ubertooth. It's as if the MCU never finds any valid executable code.

dominicgs, 2018-05-29:
Where did the
```
Checking firmware signature ........................................
........................................ ............................ Detached
```
output come from if the device never goes in to bootloader mode?  Or is that from the original firmware flashing?

dgrizzle, 2018-05-29:
It goes into bootloader mode if I short pins 1/3 as suggested. After I remove the device and plug it in again nothing happens.

dominicgs, 2018-05-29:
Could you try using that method to flash the firmware from the last release: https://github.com/greatscottgadgets/ubertooth/releases/tag/2017-03-R2

dgrizzle, 2018-05-29:
The outcome is the same. 

```
./ubertooth-dfu -d ../../../../firmware/bluetooth_rxtx/bluetooth_rxtx.dfu -r
Checking firmware signature
........................................
........................................
..................
Detached
```

dgrizzle, 2018-05-29:
@dominicgs , sorry to pepper you with messages, but when I performed with update with the 2017-03-R2 version, I used the wrong DFU file.

When using the prebuilt binaries in /ubertooth-one-firmware-bin the firmware update works as expected.

dominicgs, 2018-05-29:
Which dfu file did you use instead?

If you're building firmware from the master branch, have you got your BOARD variable set?  Do you have full build output?

dgrizzle, 2018-05-30:
I built bluetooth_rxtx from ubertooth-2017-03-R2.tar.xz. Without giving it any thought I immediately went to the bluetooth_rxtx directory and totally missed the ubertooth-one-firmware directory. 

I am not setting the board variable currently. Maybe I misunderstood but I thought it defaulted to UBERTOOTH_ONE

dominicgs, 2018-06-04:
@dgrizzle I think I need clarification on which firmware is and which isn't working.  Right now I believe that:
1) Building from git results in non-working firmware once flashed
and
2) Building from the latest release results in working firmware when flashed

Is that correct?

dgrizzle, 2018-06-04:
As of right now, no firmware built with my machine works-- does not matter if it is built from master or the 2017-03-R2 tag.

However, the bluetooth_rxtx.dfu binary included in the 2017-03-R2 release does work for me.

dgrizzle, 2018-06-05:
@dominicgs 
There appears to be something funky about the Arm embedded toolchain in the bionic repository.

Using the latest version on the arm website I was able to build from master and successfully flash my board.
```
ubertooth-util -v
Firmware version: git-d0643df* (API:1.02)
```

Download Version 7-2017-q4-major [here](https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads) 

Hopefully this helps anyone else that is having this problem.

dominicgs, 2018-06-05:
Thanks for reporting that!  I'll look at it now and see what's strange about it.

dominicgs, 2018-06-05:
Can confirm this, it's a compiler issue.  Workaround is to used 7-2017-q4-major as @dgrizzle suggests.

dominicgs, 2018-06-06:
It looks to be related to this bug: https://bugs.launchpad.net/gcc-arm-embedded/+bug/1772332

Replacing the libnewlib-arm-none-eabi and libnewlib-dev packages with the ones from Debian Sid fixes the issue.  I think this is just a case of Ubuntu needing to pick up those updated packages.

krisclarkdev, 2018-06-12:
Sorry for the slow replies, guys.  Been traveling for work.  I'll update you all this week.

mikeryan, 2020-12-05:
@krisclarkdev were you able to solve this? If you still need support, please reopen the issue.

closed 2020-12-05 by mikeryan