greatscottgadgets/ubertooth #307: Stuck on Cmake of Ubertooth tools

gregiot, 2018-05-03:
HI guys.  I'm doing the build guide, but I'm stuck on a Cmake error:
pi@raspberrypi:~/ubertooth-2017-03-R2/host/build $ cmake ..
-- Setting version string 2017-03-R2
-- Could NOT find LIBBTBB (missing:  LIBBTBB_LIBRARIES LIBBTBB_INCLUDE_DIR) 
-- Building shared library
-- Could NOT find LIBBTBB (missing:  LIBBTBB_LIBRARIES LIBBTBB_INCLUDE_DIR) 
-- Setting udev rule group to - plugdev
-- Ubertooth udev rules will be installed to '/etc/udev/rules.d' upon running 'make install'
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LIBBTBB_INCLUDE_DIR (ADVANCED)
   used as include directory in directory /home/pi/ubertooth-2017-03-R2/host/libubertooth/src
   used as include directory in directory /home/pi/ubertooth-2017-03-R2/host/libubertooth/src
   used as include directory in directory /home/pi/ubertooth-2017-03-R2/host/libubertooth/src
   used as include directory in directory /home/pi/ubertooth-2017-03-R2/host/libubertooth/src
   used as include directory in directory /home/pi/ubertooth-2017-03-R2/host/libubertooth/src
   used as include directory in directory /home/pi/ubertooth-2017-03-R2/host/libubertooth/src
   used as include directory in directory /home/pi/ubertooth-2017-03-R2/host/libubertooth/src
   used as include directory in directory /home/pi/ubertooth-2017-03-R2/host/ubertooth-tools/src
   used as include directory in directory /home/pi/ubertooth-2017-03-R2/host/ubertooth-tools/src
   used as include directory in directory /home/pi/ubertooth-2017-03-R2/host/ubertooth-tools/src
   used as include directory in directory /home/pi/ubertooth-2017-03-R2/host/ubertooth-tools/src
   used as include directory in directory /home/pi/ubertooth-2017-03-R2/host/ubertooth-tools/src
   used as include directory in directory /home/pi/ubertooth-2017-03-R2/host/ubertooth-tools/src
   used as include directory in directory /home/pi/ubertooth-2017-03-R2/host/ubertooth-tools/src
   used as include directory in directory /home/pi/ubertooth-2017-03-R2/host/ubertooth-tools/src
   used as include directory in directory /home/pi/ubertooth-2017-03-R2/host/ubertooth-tools/src
   used as include directory in directory /home/pi/ubertooth-2017-03-R2/host/ubertooth-tools/src
   used as include directory in directory /home/pi/ubertooth-2017-03-R2/host/ubertooth-tools/src
   used as include directory in directory /home/pi/ubertooth-2017-03-R2/host/ubertooth-tools/src
   used as include directory in directory /home/pi/ubertooth-2017-03-R2/host/ubertooth-tools/src
   used as include directory in directory /home/pi/ubertooth-2017-03-R2/host/ubertooth-tools/src
   used as include directory in directory /home/pi/ubertooth-2017-03-R2/host/ubertooth-tools/src
   used as include directory in directory /home/pi/ubertooth-2017-03-R2/host/ubertooth-tools/src
   used as include directory in directory /home/pi/ubertooth-2017-03-R2/host/ubertooth-tools/src
   used as include directory in directory /home/pi/ubertooth-2017-03-R2/host/ubertooth-tools/src
LIBBTBB_LIBRARIES (ADVANCED)
    linked by target "ubertooth" in directory /home/pi/ubertooth-2017-03-R2/host/libubertooth/src
    linked by target "ubertooth-debug" in directory /home/pi/ubertooth-2017-03-R2/host/ubertooth-tools/src
    linked by target "ubertooth-scan" in directory /home/pi/ubertooth-2017-03-R2/host/ubertooth-tools/src
    linked by target "ubertooth-util" in directory /home/pi/ubertooth-2017-03-R2/host/ubertooth-tools/src
    linked by target "ubertooth-dump" in directory /home/pi/ubertooth-2017-03-R2/host/ubertooth-tools/src
    linked by target "ubertooth-rx" in directory /home/pi/ubertooth-2017-03-R2/host/ubertooth-tools/src
    linked by target "ubertooth-tx" in directory /home/pi/ubertooth-2017-03-R2/host/ubertooth-tools/src
    linked by target "ubertooth-btle" in directory /home/pi/ubertooth-2017-03-R2/host/ubertooth-tools/src
    linked by target "ubertooth-afh" in directory /home/pi/ubertooth-2017-03-R2/host/ubertooth-tools/src
    linked by target "ubertooth-dfu" in directory /home/pi/ubertooth-2017-03-R2/host/ubertooth-tools/src
    linked by target "ubertooth-specan" in directory /home/pi/ubertooth-2017-03-R2/host/ubertooth-tools/src
    linked by target "ubertooth-follow" in directory /home/pi/ubertooth-2017-03-R2/host/ubertooth-tools/src
    linked by target "ubertooth-ego" in directory /home/pi/ubertooth-2017-03-R2/host/ubertooth-tools/src

-- Configuring incomplete, errors occurred!
See also "/home/pi/ubertooth-2017-03-R2/host/build/CMakeFiles/CMakeOutput.log".
pi@raspberrypi:~/ubertooth-2017-03-R2/host/build $ 

Looked at the log but it did not have any indication of what is wrong.  Help!
.

### Expected behaviour
Complete the cmake, so I can do the make
r
Tell us what happens instead

### Version information
Rasbian/Debian Stretch
Ubertooth tools version (ubertooth-rx -V):**

**libbtbb version:**

**Ubertooth firmware version (ubertooth-util -v):**

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

dominicgs, 2018-05-07:
It looks like cmake can't find libbtbb.  Assuming that you're following the build guide, have you already installed it?  Did you run the following steps after building libbtbb:
```
sudo make install
sudo ldconfig
```

mfaerevaag, 2018-12-10:
I believe this output occurs when trying to build ubertools before libbtbb, when following the guide 👍

closed 2018-12-12 by dominicgs