greatscottgadgets/ubertooth #308: qt5 support for ubertooth-specan

ZeroChaos-, 2018-05-10:
most distros are killing qt4, please update ubertooth-specan to qt5

a17r, 2018-05-16:
Official Qt4 support ended in 2015.

Etherealflux, 2018-08-19:
This caused me some grief when getting things installed - I have Python3 as default, so ubertooth was built and installed for it, but I could not get PySide to work (probably because qt4 is dead as a doornail at this point). I had to put a symlink on my path so that `python` resolved to `python2`, and then make/make install again to get `ubertooth-specan-ui` working.

A note in the build instructions about this would be helpful!

ayleph, 2018-10-25:
I submited #336 to address this issue.

dominicgs, 2018-10-25:
Fantastic, thank you for the contribution!

I've merged it.

closed 2018-10-25 by dominicgs