greatscottgadgets/ubertooth #309: Ubertooth (on Ubuntu) does not follow the BLE device with given address

Phase-angle-regulating-transformer, 2018-05-16:
Hi there

I want to sniff some data packages between two Bluetooth Low Energy Devices with Ubertooth One.

I followed steps from build guide https://github.com/greatscottgadgets/ubertooth/wiki/Build-Guide. Next I tried to sniff my devices using "ubertooth-btle -f -a(address) -c /tmp/pipe" but it does not work. The results of this operation are the same like "ubertooth-btle -f -c /tmp/pipe". Other variants of this command with -t or -s does not work too. It does not hopping between channels, only shows some addresses and infos on chosen channel.

dominicgs, 2018-05-16:
I didn't write the `ubertooth-btle` tool, so this may not be correct, but I believe you need to run it as two separate commands:
```
ubertooth-btle -t <address>
ubertooth-btle -f -c /tmp/pipe
```

mikeryan, 2018-05-16:
In the latest released firmware, you will receive advertising packets from all devices but only follow connections from the target device specified with -t.

In the le_phy branch's firmware, in addition to only following connections from the specified target device you will also only receive advertising packets to/from that device.

Phase-angle-regulating-transformer, 2018-05-17:
Thanks mikeryan. So what are the steps to follow connection from one target device? I attempt to do this too, but it failed.  Maybe I did something wrong. I used function with -t but it is still not following this device.

mikeryan, 2018-05-22:
It will likely take many tries to capture the connection. Just keep running ubertooth -f after you use -t and repeatedly disconnect and reconnect the devices. Ubertooth will eventually capture a connection. It can take 10+ tries sometimes.

closed 2018-06-27 by mikeryan