greatscottgadgets/ubertooth #310: ubertooth-btle should support filtering based on device address

Matir, 2018-05-26:
### Steps to reproduce
1. Start ubertooth-btle with a command like `ubertooth-btle -rwatch_conn2.pcap -A37 -f -tE0:A0:30:C5:E8:58`
2. Watch your screen (and pcap file) fill with packets completely unrelated to the device to be followed.

### Expected behaviour
I *expect* that it will work as described.  I would *like* it to be possible to filter to only see the advertising packets from the device, and the packets from both ends of the connection once a connection is being followed.

### Actual behaviour
You see everything (which, where I am, is an unusable mess.)

### Version information
**Operating system**: Linux human 4.14.0-3-amd64 #1 SMP Debian 4.14.17-1 (2018-02-14) x86_64 GNU/Linux

**Ubertooth tools version (ubertooth-rx -V):** libubertooth 1.0 (2017-03-R2), libbtbb 1.0 (2017-03-R2)

**libbtbb version:** 2017-03-R2

**Ubertooth firmware version (ubertooth-util -v):** ubertooth 2017-03-R2 (user@localhost)

mikeryan, 2018-05-26:
It's definitely the best firmware to use for BLE, as it's where all development is happening. It hasn't quite reached feature parity with existing master, so it hasn't merged yet. If you do find bugs in this branch, please file them!

mikeryan, 2018-06-27:
This firmware has been merged into master and is in the latest pre release 2018-06-R1.

closed 2018-06-27 by mikeryan