greatscottgadgets/ubertooth #312: What's the purpose of bluetooth-rx if we can't decrypt Bluetooth Classic packets?

SamDecrock, 2018-05-30:
Just a quick question,

I initially bought the Ubertooth One to reverse engineering the protocol of Amazon's Echo Buttons. They use regular Bluetooth Classic.

After discovering the UAP, I'm able to capture and visualize the Bluetooth Classic packets using

```
mkfifo /tmp/pipe
sudo wireshark -k -i /tmp/pipe
ubertooth-rx -l <LAP> -u <UAP> -q /tmp/pipe
```

After a while I see a packet payloads in the ubertooth-rx console, but in my Wireshark window it says `encrypted data`.

I have seen many replies by @dominicgs that decrypting Bluetooth Classic packets is not possible.

But... what's the purpose of the bluetooth-rx tool then?

dominicgs, 2018-06-04:
`ubertooth-rx` is most frequently used for finding the LAP and UAP of non-discoverable devices.

In certain circumstances, which were more common when we first wrote the tool, BT Classic devices operate unencrypted and `ubertooth-rx` was able to follow them for a time.  This just isn't as common these days, and even if we were able to capture all BT classic channels simultaneously, we could still only crack the encryption on BT 1.x devices, because 2.x use a better key exchange protocol (that I believe has not been broken yet).

SamDecrock, 2018-06-04:
All right, thanks for the explanation. I really appreciate you taking the time to answer my, probably blunt, question.

closed 2018-06-04 by SamDecrock

benmargolin, 2019-10-18:
...but what if we were able to get the symmetric key, for example on rooted Android devices it's possible to extract the key from bonds. Theoretically then we could introspect the BT classic traffic...