greatscottgadgets/ubertooth #314: Channel mapping on le_phy branch does not seem correct

gsingh93, 2018-06-03:
I'm not an expert on the Bluetooth protocol or this code so I may wrong here, but the channel mapping code does not seem correct. Specifically this line: https://github.com/greatscottgadgets/ubertooth/blob/le_phy/firmware/bluetooth_rxtx/bluetooth_le.c#L241

According to the algorithm specified in Section 4.5.8.2 Vol 6 Part B of the Bluetooth spec, we need to do `channel_idx % remapping->total_channels` on that line, otherwise we'll be reading values from an unitialized array.

mikeryan, 2018-06-03:
This does look like a bug. I had a hard time testing this code because I can't force my devices to use a partial channel map. Have you been able to reproduce this, and if so does changing the line to include a `% remapping->total_channels` fix it?

gsingh93, 2018-06-03:
I haven't tried to apply that fix yet, but I did test out the unmodified `le_phy` branch and it doesn't work well for me, so there is definitely some bug here. The connect packet seems like garbage and other packets don't look right either. I'll try to test this and see if packet captures look more normal, but if they still look strange it might be another bug.

gsingh93, 2018-06-03:
Yea, so I applied the fix but I don't think I'll be able to test until https://github.com/greatscottgadgets/ubertooth/issues/315 is solved. Let me know if I can help, I'm not sure about the best way to start debugging this.

closed 2018-06-19 by mikeryan