greatscottgadgets/ubertooth #315: CONNECT_REQ packet is incorrect on le_phy branch

gsingh93, 2018-06-03:
On 2017-03-R2 CONNECT_REQ packets show up correctly, but with the le_phy branch they're wrong.

Take a look at this correct CONNECT_REQ:

[image]

As you can see the values are normal. Now with the same device, I capture a connection using the `le_phy` firmware:

[image]

These values are not normal.

Let me know if there's anything I can do to help with debugging this.

mikeryan, 2018-06-03:
Can you provide any details on the BLE devices involved? Is the system running Ubertooth an x64 Linux system?

gsingh93, 2018-06-03:
@mikeryan The system running Ubertooth is Ubuntu 16.04 64-bit. The BLE device is an Oculus Go headset.

mikeryan, 2018-06-03:
This is really unusual. I've been using the le_phy branch on a regular basis and the CONNECT_REQ packets look fine to me. Are you sure the packet you're looking at wasn't the victim of random RF noise? Are you seeing zero cases of valid packets?

Would be helpful if you could provide PCAP files from both master and le_phy. Also make sure you're using libbtbb master and build/install the host tools from le_phy.

gsingh93, 2018-06-04:
> Are you sure the packet you're looking at wasn't the victim of random RF noise?

While it does feel like it takes more tries to capture a connect with le_phy, I only see CONNECT_REQ when I attempt to pair. If I leave it running for a while, I don't see random CONNECT_REQ packets. So I don't think this is RF noise.

> Are you seeing zero cases of valid packets?

I'm not familiar enough with BLE to know how bad the trace is. However, I do know that SCAN_RSP packets are also not correct. If you look at the SCAN_RSP packets send from the OculusVr device, you'll see the capture from the master branch looks different from the le_phy branch (I know for sure the SCAN_RSP on master is correct)

You can find two captures here: [captures.zip](https://github.com/greatscottgadgets/ubertooth/files/2066644/captures.zip)

`master.pcap` is a using the master branch for libbtbb, the host tools, and the firmware.
`le_phy.pcap` is using the master branch for libbtbb, and the le_phy branch for the host tools and firmware. I wasn't able to get the le_phy_wip branch of libbtbb working.

Notice how the CONNECT_REQ packet in `le_phy.pcap` is definitely wrong. Also notice how the SCAN_RSP packets differ. I'm also seeing a lot of L2CAP fragment packets after connect, but I'm guessing that's because partial channel mapping isn't working.

Let me know if you need anything else.

mikeryan, 2018-06-20:
le_phy is merged into master along with numerous bug fixes (including partial channel map support). Please try master's firmware and host tools and see if that fixes things up.

mikeryan, 2018-06-27:
Please try the latest git master, and make clean before building the firmware.

gsingh93, 2018-07-02:
Still no luck, lots of malformed packets and the CONNECT packets still look bad.

mikeryan, 2020-12-05:
@gsingh93 many improvements have been made to the firmware since you opened this issue. Please try the latest master or next release and reopen the issue if you still find you have the problem.

closed 2020-12-05 by mikeryan