greatscottgadgets/ubertooth #316: ubertooth-btle -s __ what is it doing?

muphenmahn, 2018-06-05:
I know the transmit function of this btle is experimental, but i do have a question about trying to understand it. When i use the command, ubertooth-btle -s 11:33:55:99:BB:DD, I'm on kali btw, i see the transmit light turn on. the -s option is for "Send advertising packets". What should i see when this is going on, should my phone detect a new access point to connect to? or is this something entirely different, i had a hard time understanding what an advertising packet is, I'm trying to transmit a fake Bluetooth access point, not sure if I'm using the right tool. Thanks a lot.

mikeryan, 2018-06-05:
This tool will transmit fake ADV_IND packets for the specified BD address. Using a BLE scanner app on your phone or laptop (such as LightBlue on iOS or OS X, or hcitool lescan on Linux) you should see the fake BD address in the device list.

I'm not sure what you mean by "Access Point", as this does not have a lot of meaning in the context of Bluetooth.

muphenmahn, 2018-06-06:
roger that man, that makes perfect sense, thanks for the reply. And by access point i meant the broadcast SSID of a bluetooth device such as "bose headphones - 334" you know? Its a close enough meaning lol, hopefully you smell what im cooking. Could your code possibly transmit that SSID? or is it even possible? just looking for some help.

mikeryan, 2018-06-27:
You will have to manually construct an ADV_IND packet with the appropriate AD data and modify the firmware in order to send it. There is no userspace support for generating such packets at the moment. Take a look at the advertising packets generated by a pair of headphones using Ubertooth + Wireshark to get an idea of what that looks like.

closed 2020-12-05 by mikeryan