greatscottgadgets/ubertooth #319: No longer able to upgrade FW

eblot, 2018-07-03:
### Steps to reproduce
0. I've tried the following sequences on both Debian 9.3 and on macOS 10.13.5
1. Download `libbtbb` from Git, build, install
2. Download `ubertooth-2018-06-R1.tar.xz` (also tried with Git master)
 
   * Build the host tools, install, build the FW

3. Attempt to reflash
   
       sudo ubertooth-dfu -r -u ./bluetooth_rxtx/bluetooth_rxtx.dfu

       Switching to DFU mode...
       ... (....)
       Detached

or

       ... (....)
       control message unsupported

4. Unplug/replug the Ubertooth one, also ran `sudo ubertooth-util -r` about 20 times and try to reflash again (USB VID/PID move back and forth from `1d50:6002` to `1d50:6003` depending on the device's state.

5. Read the version (each time after upgrade attempt), always get

       sudo ubertooth-util -v

       Firmware version: 2017-03-R2 (API:1.02)
    
### Expected behaviour

I expect the DFU tool to show the documented string:

    Checking firmware signature

then flash the FW, and get a new FW version to to show up (2018-06-... ?)

I'm not sure what to do next to upgrade the FW with the new release.

### Actual behaviour

Seems the device cannot be upgraded anymore.

### Version information
**Operating system**:
  * Debian 9.3 (4.9.0-6-amd64 #1 SMP Debian 4.9.88-1+deb9u1 (2018-05-07) x86_64 GNU/Linux)
  * macOS 10.13.5 (17.6.0 Darwin Kernel Version 17.6.0: Tue May  8 15:22:16 PDT 2018; root:xnu-4570.61.1~1/RELEASE_X86_64 x86_64)

**Ubertooth tools version (ubertooth-rx -V):**

  libubertooth 1.1 (git-a6332a7), libbtbb 1.0 (git-d615b97)

**Ubertooth firmware version (ubertooth-util -v):**

  ubertooth 2017-03-R2 (dominicgs@hydrogen) Mon Mar 13 16:09:02 MDT 2017

eblot, 2018-07-03:
Me, stupid: `download` is seen from the device perspective, so upload is the read direction, and download is the write direction. `-d` does work.

closed 2018-07-03 by eblot

dominicgs, 2018-07-03:
Sorry about that confusion, we use the terms from the DFU specification, but I agree that they are rather ambiguous.